artisan-roaster-scope/artisan #643: Permission Errors

NapTight, 2021-07-17:
## Expected Behaviour
Import device temps using prog feature

## Actual Behaviour
Receiving error:
Exception: callprogram(): [Errno 13] Permission denied: '/home/pi/RPiSensors/beantemp.py' @line 690

## Steps to Reproduce the Problem

  1. select Device from Config menu
  2. select prog from ET/BT tab
  3. use select button to find python3 beantemp.py program and select it 

## Specifications

  - Artisan Version: 2.4.4
  - Artisan Build (number in brackets shown in the about box): 0
  - Platform (Mac/Windows/Linux + OS version): linux raspbian buster
  - Connected devices or roasting machine: rpi4b + max31855 + thermocouple
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Settings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._
[beantemp.txt](https://github.com/artisan-roaster-scope/artisan/files/6835273/beantemp.txt)
[factory-artisan-settings.txt](https://github.com/artisan-roaster-scope/artisan/files/6835275/factory-artisan-settings.txt)

MAKOMO, 2021-07-20:
Obviously the Artisan process has not the rights to start your script. Note that your script is a python script, but Artisan is expecting to call shell script. You need to make sure that your script can be executed from the shell, so it might need require some first line telling it that the content should be handed over to a Python process (like `#!/usr/bin/env python3`).

I close this as this is obviously not an Artisan issue.

closed 2021-07-20 by MAKOMO

NapTight, 2021-07-20:
Thanks for your reply. Do you know where I can find scripts that work with Artisan? I added the shebang you offered, but I'm still having the same issue. The script works in terminal if I call it with $ python3 beantemp.py but not if I call it directly with ./beantemp.py

MAKOMO, 2021-07-21:
Your script works as you call python to interpret your Python script. However, Artisan is just calling your script, so if your script without feeding it into Python and can be run just by the shell itself you are good to go:

```
$ ./beantemp.py
```

NapTight, 2021-07-21:
Thank you. This help me figure it out. sorry about bugging you in the
Artisan page. Thank you for maintaining it!