artisan-roaster-scope/artisan #652: Exception: fireslideraction() in 2.4.6

DavidFW1960, 2021-07-30:
**Describe the bug**
A clear and concise description of what the bug is.
Following on from:
https://github.com/artisan-roaster-scope/artisan/issues/558
The new 2.4.6 has a different but related issue I think.
**To Reproduce**
Steps to reproduce the behavior:
I am using the same alarms as I was in the previous posted issue. 
I get that error '22 10:41:36.589 Exception: fireslideraction() expected string or bytes-like object@line 21527' every time an alarm is called and the alarm doesn't do what is expected (Vary the heat or fan speed)
Here is the log
```
version = 2.4.6

4 11:26:38.040 Exception: fireslideraction() expected string or bytes-like object@line 21527

3 11:26:38.040 Exception: fireslideraction() expected string or bytes-like object@line 21527

2 11:22:31.776 Exception: fireslideraction() expected string or bytes-like object@line 21527

1 11:22:11.769 Exception: fireslideraction() expected string or bytes-like object@line 2152776 11:27:16.430 Event # 7 recorded at BT = 144.0 Time = 09:24

75 11:27:09.571 Event # 6 recorded at BT = 143.9 Time = 09:18

74 11:26:40.356 Exception: fireslideraction() expected string or bytes-like object@line 21527

73 11:26:37.509 Event # 5 recorded at BT = 139.9 Time = 08:46

72 11:26:37.506 Alarm 2 triggered

71 11:26:35.604 Event # 4 recorded at BT = 139.2 Time = 08:44

70 11:26:35.601 Alarm 3 triggered

69 11:22:34.373 Exception: fireslideraction() expected string or bytes-like object@line 21527

68 11:22:31.492 Event # 3 recorded at BT = 103.0 Time = 04:40

67 11:22:31.489 Alarm 11 triggered

66 11:22:14.356 Exception: fireslideraction() expected string or bytes-like object@line 21527

65 11:22:11.499 Event # 2 recorded at BT = 99.6 Time = 04:20

64 11:22:11.495 Alarm 1 triggered

63 11:19:11.419 [TP] recorded at 1:12 BT = 61.0 C

62 11:17:56.907 Roast time starts now 00:00 BT = 99.7C

61 11:17:56.331 Event # 1 recorded at BT = 94.2 Time = 00:04

60 11:17:38.206 Alarm 9 triggered

59 11:16:55.975 Alarm is calling: ./home/pi/artisan_alarm.sh

58 11:16:55.970 Alarm 12 triggered

57 11:13:54.216 Scope recording...

56 11:13:33.485 Scope monitoring...

55 11:13:30.323 Scope has been reset

54 11:13:26.686 Scope has been reset

53 11:13:20.499 Scope stopped

52 11:13:20.495 PID OFF

51 11:13:12.162 Scope recording stopped

50 11:12:55.739 Alarm is calling: ./home/pi/artisan_alarm_clear.sh

49 11:12:55.292 Alarm 14 triggered

48 10:51:48.313 [COOL END] recorded at 10:56 BT = 174.3 C

47 10:51:48.309 Alarm 10 triggered

46 10:46:47.601 Roast ended at 05:56 BT = 202.6 C

45 10:46:44.599 Alarm 4 triggered

44 10:46:42.690 Alarm is calling: ./home/pi/artisan_alarm_end.sh

43 10:46:42.338 Alarm 13 triggered

42 10:45:55.787 Exception: fireslideraction() expected string or bytes-like object@line 21527

41 10:45:55.510 Event # 15 recorded at BT = 189.8 Time = 05:02

40 10:45:53.770 Event # 14 recorded at BT = 189.8 Time = 05:02

39 10:45:53.744 Alarm 7 triggered

38 10:45:49.538 Exception: fireslideraction() expected string or bytes-like object@line 21527

37 10:45:49.256 Event # 13 recorded at BT = 188.2 Time = 04:56

36 10:45:47.756 Event # 12 recorded at BT = 188.2 Time = 04:56

35 10:45:47.753 Alarm 8 triggered

34 10:45:16.583 Exception: fireslideraction() expected string or bytes-like object@line 21527

33 10:45:13.740 Event # 11 recorded at BT = 179.7 Time = 04:22

32 10:45:13.736 Alarm 6 triggered

31 10:44:26.590 Exception: fireslideraction() expected string or bytes-like object@line 21527

30 10:44:23.761 Event # 10 recorded at BT = 164.6 Time = 03:32

29 10:44:23.758 Alarm 5 triggered

28 10:43:02.869 Exception: fireslideraction() expected string or bytes-like object@line 21527

27 10:43:01.822 Event # 9 recorded at BT = 140.9 Time = 02:10

26 10:43:01.814 Alarm 2 triggered

25 10:43:01.811 Exception: fireslideraction() expected string or bytes-like object@line 21527

24 10:42:57.716 Event # 8 recorded at BT = 139.6 Time = 02:06

23 10:42:57.712 Alarm 3 triggered

22 10:41:36.589 Exception: fireslideraction() expected string or bytes-like object@line 21527

21 10:41:33.714 Event # 7 recorded at BT = 116.5 Time = 00:42

20 10:41:33.711 Alarm 11 triggered

19 10:41:23.179 Event # 6 recorded at BT = 112.7 Time = 00:30

18 10:41:21.477 Event # 5 recorded at BT = 112.7 Time = 00:30

17 10:41:16.594 Exception: fireslideraction() expected string or bytes-like object@line 21527

16 10:41:14.827 Event # 4 recorded at BT = 109.8 Time = 00:22

15 10:41:14.822 Alarm 1 triggered

14 10:41:14.819 [TP] recorded at 0:00 BT = 100.9 C

13 10:40:56.211 Event # 3 recorded at BT = 100.9 Time = 00:00

12 10:40:54.618 Event # 2 recorded at BT = 100.9 Time = 00:00

11 10:40:53.526 Roast time starts now 00:00 BT = 100.9C

10 10:40:53.521 Event # 1 recorded at BT = 100.9 Time = 00:00

9 10:40:44.428 Alarm 9 triggered

8 10:40:15.417 Alarm is calling: ./home/pi/artisan_alarm.sh

7 10:40:15.302 Alarm 12 triggered

6 10:34:50.825 Scope recording...

5 10:34:37.558 Hottop control turned on

4 10:34:34.815 super on

3 10:34:29.728 Scope monitoring...

2 10:34:26.635 Scope has been reset

1 10:34:08.901 
```

**Expected behavior**
A clear and concise description of what you expected to happen.
2.4.2 works as expected. 2.4.4 completely locks up and freezes the system as per the previous issue. In 2.4.6 it shows the slider on the left with the correct heat/fan but the LEDs are not showing the right data and the roaster isn't changing it's settings either.
**Screenshots**
If applicable, add screenshots to help explain your problem.

**Setup (please complete the following information):**
 - Artisan Version [e.g. v2.4.4]:  2.4.6
 - Artisan Build (number in brackets shown in the about box) [e.g. 97e6dec]: 31184b1b
 - Version [e.g. 22]: 1.7?
 - Platform (Mac/Windows/Linux + OS version) [e.g. macOS 10.15.3]: Raspbian
 - Connected devices or roasting machine [e.g. Probatone 5]: Hottop
[artisan-settings.txt](https://github.com/artisan-roaster-scope/artisan/files/6904822/artisan-settings.txt)
[profile.alrm.txt](https://github.com/artisan-roaster-scope/artisan/files/6904823/profile.alrm.txt)

[image]
Screenshow shows slider and LED's different.

**Additional context**
Add any other context about the problem here.
At least it's not locking up the roaster and I can manually use buttons on the bottom to set the heat and fan speed..
Please attach your current Artisan settings file (as exported via menu Help >> Save Settings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2021-07-30:
Dear David, thanks for this detailed report. I could easily reproduce this. Indeed the initial release of v2.4.6 is breaking all Hottop (and Aillio) slider actions (not only if triggered via alarms, but also if triggered directly by moving the sliders). The fix was easy. At one point an integer was forwarded instead of a string containing that integer. I just fixed and triggered new builds. It will take us a moment to update the builds to revision v2.4.6.1. This fix does not resolve Issue #558 as in v2.4.2 the same issue existed, it just did not manifest as it was somehow correct later in the code. Still I have some hopes that other changes applied to v2.4.6 might resolve #558 (which I still cannot reproduce).

I will ping you once the new builds are online.

DavidFW1960, 2021-07-30:
Thanks Marko. I'm optimistic that the 2.4.4 issue might be fixed because it didn't lock me up like 2.4.4 was doing so I'll see what happens when you release it. 2.4.2 is working perfectly..

MAKOMO, 2021-07-30:
Dave, the updated builds are online now. Please re-download and install again. This one is build 1 while the original build was build 0 (check the About box). Hope you can confirm that this fixes the slider issue for the Hottop. I still keep the Issue #558 open as the failure of this issue might have just blocked the hang/crash in #558. I still think that that one is platform related as on the RPi we have to work with outdated Qt libs. The next Raspian release will update to more current Qt libs.

DavidFW1960, 2021-07-30:
Thanks Marko. I'll grab it tomorrow and see

MAKOMO, 2021-07-30:
No need to roast or to check the alarms. Just connect and move the sliders and see if they work.

DavidFW1960, 2021-07-31:
OK definitely fixes slider and alarms etc all working. So far with no beans and taping charge it hasn't locked up either. I'll do a real roast when the test completes.... Fingers crossed...

MAKOMO, 2021-07-31:
Good news Dave! I will close this one and hope to be able to close the other issue after you did a row of successful roast over the next days. Thanks for this excellent detailed report and for testing the fix.

closed 2021-07-31 by MAKOMO

DavidFW1960, 2021-07-31:
I did one roast and it was fine the second I don't know what happened - it didn't appear to lockup... but I ended up with a fire in the roaster because I wasn't watching it properly and now it won't turn the heater on so it will be a few weeks I think before I can check again.....

MAKOMO, 2021-07-31:
Oh, I am sorry to hear that. Hopefully it is only the heating element, which can be replaced quite easily. Good that the fire was only in the roaster and nobody got injured. Hope you can resolve this easy enough.