artisan-roaster-scope/artisan #656: ET not detected

cbc1121, 2021-08-05:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Setup (please complete the following information):**
 - Artisan Version [e.g. v2.4.4]: 
 - Artisan Build (number in brackets shown in the about box) [e.g. 97e6dec]: 
 - Version [e.g. 22]: 
 - Platform (Mac/Windows/Linux + OS version) [e.g. macOS 10.15.3]: 
 - Connected devices or roasting machine [e.g. Probatone 5]: 

**Additional context**
Add any other context about the problem here.

Please attach your current Artisan settings file (as exported via menu Help >> Save Settings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

roasterdave, 2021-08-05:
@cbc1121 It is not possible to help when you provide no information.  Please update with the information requested in the issue form above.

cbc1121, 2021-08-06:
> @cbc1121 It is not possible to help when you provide no information. Please update with the information requested in the issue form above.

hi dave,
I edited my post :)

MAKOMO, 2021-08-06:
You seem to have a Santoker BT USB (eg. R3) machine which communicates via USB using the Center 301 protocol. However, for some reasons you have a mixed setup which blends a MODBUS extra device in. 

To fix:

1) menu Config >> Device, 2nd tab "Extra Devices", press Reset to remove that one extra device
2) menu Config >> Device, 3rd tab "Symb ET/BT", remove the "Y3" from the "ET Y(x)" field
3) menu Config >> Port, 3rd tab "MODBUS", set all Slave IDs in the first row of the table to 0

closed 2021-08-06 by MAKOMO

cbc1121, 2021-08-07:
Thanks Marco. Problem solve 🤟🏾

roasterdave, 2021-08-07:
Excellent!

cbc1121, 2021-10-01:
Hi,

My artisan go something off again. When I’m not roasting. The BT difference number keep going up and down. 

Here video when I’m not roasting