artisan-roaster-scope/artisan #657: Ability to configure default save filename template

edgimar, 2021-08-06:
When saving a roast, the current default filename format is "<title>_<date>".  It would be nice if this were configurable, so that you could choose a format (e.g. something like "%Y%m%d%H%M_%{title}") that would be used to generate the default filename.

CarefreeBuzzBuzz, 2021-08-06:
> When saving a roast, the current default filename format is "<title>". It would be nice if this were configurable, so that you could choose a format (e.g. something like "%Y%m%d%H%M%{title}") that would be used to generate the default filename.

Please look at the Quick Start Guide section on Autosave. 

Michael Herbert 
CarefreeBuzzBuzz

MAKOMO, 2021-08-06:
There is also a whole block post on this sophisticated mechanism: [Autosave File Naming](https://artisan-roasterscope.blogspot.com/2020/05/autosave-file-naming.html)

closed 2021-08-06 by MAKOMO