artisan-roaster-scope/artisan #667: Issues with Batch Counter

stuartmcknight, 2021-08-17:
Enabled Batch Counter in configuration. Say set to 253 as initial count.

**To Reproduce**
When creating a new roast, from the title bar it appears that the batch number is correctly displayed using ~batch as 253 with the rest of the roast properties,

However, once roast completes, the Autosave has now changed the 253 to 232. And when reopening the Batch dialog, the new seed number seems to have DECREMENTED to 231. Artisan has not been restarted during these runs so I don't believe it has reloaded the config (although the don't overwrite checkbox is enabled anyway)
**Expected behavior**
Expected batch number to correctly be saved with alog file and counter to increment +1 as the dialog box implies.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Setup (please complete the following information):**
 - Artisan Version [e.g. v2.4.4]: 2.4.6
 - Artisan Build (number in brackets shown in the about box) [e.g. 97e6dec]: 
 - Version [e.g. 22]: 
 - Platform (Mac/Windows/Linux + OS version) [e.g. macOS 10.15.3]: Windows 10 
 - Connected devices or roasting machine [e.g. Probatone 5]: Aillio Bullet

**Additional context**
Add any other context about the problem here.

Please attach your current Artisan settings file (as exported via menu Hel
[Archive.zip](https://github.com/artisan-roaster-scope/artisan/files/7001610/Archive.zip)
p >> Save Settings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

roasterdave, 2021-08-17:
Please do not duplicate issues and discussions.  Best to keep all the information in a single place. 

Sorry I do not understand "When creating a new roast, from the title bar it appears that the batch number is correctly displayed using ~batch as 253 with the rest of the roast properties,"  Can you provide either a screenshot or a step-by-step description showing how to reproduce the problem here?

The batch number shown in roast properties will be used by Autosave at the time the file is saved.  

I tested using your settings file and could not find any problem with the operation of the batch counter.  Again, providing a step-by-step procedure to reproduce the problem will help us to go further to see what is the issue.

stuartmcknight, 2021-08-18:
Sorry Dave I did search for  “batch” and for whatever reason there were no hits.

I will see if I can add relevant screen shot later but to reiterate steps

Autosave and Batch are both enabled. Auto save is using the ~batch variable to create the file name with other parameters. It is the first item in the list. 
So let’s say I go into the Batch dialog and the current value is 253. I close and start a new roast filling in the profile name. At the top of the graph, there is now SC253….blah blah blah. 

I run through the roast, recording events including Drop.

when I hit Stop however, I see the profile was saved but with “SC232…blah blah.alog “. Confirmed on the file system that this matched the name. The legend above the graph also changed to SC232. 

More bizarre if I then go back into Batch dialog, the “current” counter is now 231. Not 233 as would be expected as it should be incremental not decrement. 

I have restarted Artisan multiple times, same results.

stuartmcknight, 2021-08-18:
I also tried disabling and re enable Batch but no change..

poundy, 2021-08-18:
you mention 231 and 253. Did you blow thru 20 roasts and it went backwards each time, or was one of those a mistype? 

I'm guessing but think the way that we'll figure this out will be to look at your settings file from a point in time (pre first roast) and a then look at each of your next two roast files, and the post-second roast settings file, so that we can see the changes that are being made and how things are recorded.   @roasterdave will likely comment more if he needs more than that.

poundy, 2021-08-18:
BTW I am running 2.4.6 on Win10 with batch and autosave working fine, I've not that long ago moved to a modbus roaster so did massive config changes, but didn't see any issue like you've reported. My autosave name is ~batch-~date_long_~time_~beans_25_~weight so you can see it'd also affect me if it was a bug; in my mind it'll be a setting you have so your config file will be important to investigate

stuartmcknight, 2021-08-18:
@poundy no, just one roast took me back from 253 to 232. That’s the damndest thing. Wondered if I had stumbled on some magic hexed number (close to 254!). It seems whatever I change the seed number to, it reverts to 232 and going back into the Batch dialog, it is one less at 231. 

I did attach my current settings. I guess I could reset back to factory defaults but that would be painful to recreate everything…

poundy, 2021-08-18:
you closed this ? I wouldn't, it still makes sense to investigate. It really does seem strange that it behaves like this.

Just reviewing your settings, the setting I have is:
neverUpdateBatchCounter=0
where you have it as:
neverUpdateBatchCounter=true

Is that something you just tried changing?

stuartmcknight, 2021-08-18:
Sorry can’t even work a web browser now!

I did change that setting yes to see if it made any difference but didn’t change it back before opening this..

I can only think there’s something strange in my setup but what…

I will try a few more experiment to see if I can spot a pattern…

poundy, 2021-08-18:
lol - no worries :) 
I'd suggest you correct the never update setting, and then copy the settings again as the "before" version, then do two roasts in your "usual" process, then save the settings again ("after"). Send those back and we'll all look.

roasterdave, 2021-08-18:
Please do report back if you are able to spot how /when this happens.  Try as I have I can't reproduce it on Windows 10.

stuartmcknight, 2021-08-19:
Hi guys, I think I've isolated the repro but I don't know what is broken. Here's what I experimented with tonight -

Saved my Artisan settings out as a backup then did a Factory Reset, went into Configure Device and selected my machine Aillio Bullet IBTS. Tested a couple of dummy roasts, problem still there.
Experimented with disabling AutoSave and removing the prefix from the Batch config - no change. An initial value of 1000 in the Batch resulted in a 237 value when hitting STOP. Definitely marked DROP.

In case it was a Windows 10 thing, pulled over my Macbook Pro running Catalina and loaded same 2.4.6 - again configured my device as Aillio IBTS. Same exact symptoms, so not machine and not Artisan settings.

Purely by accident, on another Mac, I started up v2.4.6 but with no device configured. Guess what. Batch now works as described. Hmmm. Configured Artisan against an imaginary Behmor from the device list. It moaned about not being able to communicate but Batch worked as described.

Went back to my roasting laptop running Windows 10, v2.4.6 - reset the factory defaults and did NOT configure any device, Batch works. Chose dummy Behmor as device. Batch works. Selected Aillio (non IBTS this time). Problem returns. Batch number seems to get stuck at 237.

So, it would seem there is something broken in the Aillio code which is affecting Batch. Maybe some value coming in over the USB which is being misread - IDK.

I did roll back through versions 2.4.4, 2.4.2, 2.4.0 configured against Aillio and the problem remained. So I don't know when the wheels came off as I only recently started to think about using Batch.

Hope this helps, but it explains why yours works, maybe the number of Aillio Artisan users working with Batch is a lot rarer than we could imagine.

MAKOMO, 2021-08-19:
Looked in the code for the R1 and it set the batch counter to the "roastNumber" it receives from the connected R1. I don't know anything about that machine, but I assume that the R1 increases this number on completed roasts. Maybe it is even possible to manual adjust this number on the machine.

MAKOMO, 2021-08-19:
Oh, actually it seems that it takes the "roastNumber" as batch number only on loading a profile from RoastWorld. On just doing a roast the R1 code in Artisan seems not to touch the batch counter at all. This needs further investigation. Still not able to reproduce this.

stuartmcknight, 2021-08-19:
Hi Marko

RoastWorld is the online "backup" of Aillio's Roast Time software which runs on the connected computer - pretty much an Artisan.plus and Artisan analogy. There is an Info panel in Roast Time which pulls up some information like the supply voltage, firmware version of the R1, temperatures on the PCB induction coils etc. There is also a number of roasts count but that is currently in the 500's and is not changeable from the software. I'm guessing it's like a lifecycle count on a battery or the odometer on a car to show how "used" it is.

I did try a quick test on my Mac with R1 set as the machine but not actually connected and got the warning about not being able to connect but the batch count did work. I know Artisan surfaces some data like bean probe temperatures, drum speed, fan speed, heater and supply voltage. Is there another value coming in from something else that is being misread as a roast count ? Does the Behmor do the same for it definitely doesn't keep track of roasts and out of the box I don't think it can even track heat and it has no fan. To instrument a Behmor involves drilling holes to mount probes so there's no contact as such with any firmware in the machine....

I wasn't even aware that Artisan could talk to Roast World as that would need some credentials just like Artisan plus and theres nowhere in Artisan to provide those, certainly not on the R1 panel itself. 

I will see if I can pull in a favour or two from one of the Aillio devs. Is there a link to the R1 driver code that I could forward ?

stuartmcknight, 2021-08-19:
I had a quick look at some of the Aillio code files and I think Artisan is getting roast count from either the R1 or somewhere in Roast World which seems strange to me. Shouldn't Artisan just be managing batch count stuff locally ? Roast World has no provision for prefixes for example. I checked my roast count on Roast World and it is currently 232. I don't know if you had seen references to "state.roast_number" in your earlier comments/diagnosis.

I'm assuming the reason Batch works when Aillio R1 is configured as the machine but it is not present on the USB bus is because the code that would be called if it were, does not get executed ?

stuartmcknight, 2021-08-19:
I'm not sure why src/artisanlib/aillio.py is trying to parse JSON files or reach out to Roast World for any of the Artisan things. Roast World has been revised in a number of ways since 2018 so I doubt these would even work any more. It really only should deal with the temperatures, fan and heat settings, mode (preheat/roasting/cooling/off) and the exhaust fan speed and voltage. It would be nice if the IBTS fan speed was available but not essential.

I can supply a sample of the JSON files that I found on my local system if it helps - I do see a "roastNumber" variable in there. But Aillio may change all that at any time of course. Also assumptions about Roast World are risky - I know they are working on a new security model.

It would seem that this code tried to be clever and link into Aillio's own software which seems pointless in the context of the Batch counter function of Artisan. I appreciate you don't have an R1 to play with/test, but I would be happy to run a special version if you can remove any attempts to get roastNumber (or equivalent)

This seems to be likely place to remove

 try:
            if "roastNumber" in data:
                res["roastbatchnr"] = int(data["roastNumber"])
        except Exception: # pylint: disable=broad-except

LMK if you need any additional data....

MAKOMO, 2021-08-19:
Those lines are only executed on importing from RoastWorld via menu File >> Import. Otherwise the R1 module is not accessing the batch counter at all. Thus there is nothing special on using the R1 module vs every other device setup regarding the batch counter.

stuartmcknight, 2021-08-19:
I realized that after digging around. But plainly there IS something different when the R1 is connected to USB when it fails but when set up as R1 but NOT connected it works - this block seems to be getting a **roast_number** inbound - does that end up overwriting the Batch counter ?

        try:
            usb.util.claim_interface(self.usbhandle, self.AILLIO_INTERFACE)
        except Exception as e:
            self.usbhandle = None
            raise IOError("unable to claim interface") from e
        self.__sendcmd(self.AILLIO_CMD_INFO1)
        reply = self.__readreply(32)
        sn = unpack('h', reply[0:2])[0]
        firmware = unpack('h', reply[24:26])[0]
        self.__dbg('serial number: ' + str(sn))
        self.__dbg('firmware version: ' + str(firmware))
        self.__sendcmd(self.AILLIO_CMD_INFO2)
        reply = self.__readreply(36)
        self.roast_number = unpack('>I', reply[27:31])[0]
        self.__dbg('number of roasts: ' + str(self.roast_number))
        self.parent_pipe, self.child_pipe = Pipe()
        self.worker_thread = threading.Thread(target=self.__updatestate,
                                              args=(self.child_pipe,))
        self.worker_thread.start()

stuartmcknight, 2021-08-20:
In comm.py ?

try:
            if self.aw.qmc.batchcounter != -1:
                self.aw.qmc.batchcounter = self.R1.get_roast_number()

stuartmcknight, 2021-08-21:
I managed a quick Slack chat with one of the Aillio folk who said there was never a public document on the array format that comes back from the R1 and over the course of time some of those have changed, including the roast_number in the code block above. Most consistently I have been getting 236 displayed in the roaster scope with the Batch seed count showing 235.

I have recordings of the USB traffic if that is helpful in lieu of having an R1 to test with. But if this code was based on reverse engineering then it would certainly be wise to throw away roast count and probably serial number and firmware version, I don’t believe that surfaces anywhere in Artisan UI.

Ive got a request in for the details of the current array format but as the temperatures and fan and drum and power settings are accurate, there’s no need I think to wait for this. Aillio claim that they are working towards an API which would help everyone going forward….

stuartmcknight, 2021-08-21:
And just to reiterate, it all works until you have an R1 connected….if Artisan is configured against an Aillio but you don’t have one plugged in, you won’t see the issue…

roasterdave, 2021-08-21:
Thanks for this information.  The batchcounter update code you found in comm.py has been removed.  There is a [continuous build](https://github.com/artisan-roaster-scope/artisan/releases/tag/continuous) available with this change.  Caution, the continuous builds may have new issues not found in release v2.4.6.  Please let us know if this resolves the problem with the R1 and batchcounter.

If Aillio publishes an API it will make it possible for Artisan to support the R1 more completely.  R1 users need to continue to ask Aillio to provide this information and interfaces necessary for Artisan to operate.  Thanks for making USB captures.  They might be useful in the future.  At this time, however, there is no resource available in the team to analyze and modify the R1 interface code.  We do welcome pull requests from other interested contributors.   

Thanks again for your investigations!

stuartmcknight, 2021-08-21:
Hi Dave,

I was idly thinking the other day how cool it might be to have say a Raspberry Pi Roaster emulator that could be used for testing ;-)

I will of course try to keep Aillio honest and get documentation into the public domain.

But ta-da your surgical removal appears to have done the trick. The Batch now works as described and there don't appear to be any R1 downsides - fan, drum, power and temperatures seem untouched. 

Thanks so much.

Stuart

roasterdave, 2021-08-21:
Great to hear this resolved the batch counter for you.  I will close this issue.  Now is a good time to use the Discussions for other topics that are not specific issues.

Emulators for physical devices are tricky.  There can be are all sorts of timing concerns.  To build a good emulator you need to know the intimate details of the interface logic, which is a missing piece with the R1.  Even with a "standard" interface there are all sorts of gotchas where the device does not behave exactly as expected (or emulated).

closed 2021-08-21 by roasterdave