artisan-roaster-scope/artisan #670: Building on mac OS X M1 processor seems to be finnicky.

benwah, 2021-08-18:
**Describe the bug**

Trying to build Artisan Roaster Scope on a macbook with M1 processor fails.

System specs:

[image]

**To Reproduce**

1. Clone project with git
2. pip install -r src/requirements.txt
3. pip install -r src/requirements-osx.txt

This will result in compilation errors when building scipy:

```
the clang compiler does not support 'faltivec', please use -maltivec
```

Similar to https://github.com/numpy/numpy/issues/17807

**Seems like there's a workaround for installing scipy**

```
OPENBLAS="$(brew --prefix openblas)" pip install scipy==1.7.0
```

CarefreeBuzzBuzz, 2021-08-18:
> Similar to numpy/numpy#17807 <https://github.com/numpy/numpy/issues/17807>

You need to run it with the Intel emulator.  No M1 support.  
----------------------------------------
Michael Herbert, Head Roaster
Order at www.carefreebuzzbuzz.com <http://www.carefreebuzzbuzz.com/>
Artisan RoasterScope  <https://artisan-scope.org/>Team

benwah, 2021-08-18:
Ok workaround so far:

1. Install openblas and then, scipy, manuallt:

```
brew install openblas
OPENBLAS="$(brew --prefix openblas)" pip install scipy=1.7.0
```

2. Install QT5. I used Homebrew:

```
brew install qt@5
```

3. Link the binaries to have access to `qmake` and such things.

```
brew link qt5 --force
```

4. May or may not have to do this step, set some session variables:

```
export LDFLAGS="-L/opt/homebrew/opt/qt@5/lib"
export CPPFLAGS="-I/opt/homebrew/opt/qt@5/include"
```

5. Install pip requirements:

```
pip install -r src/requirements.txt
pip install -r src/requirements-osx.txt
```

... PyQt5 is now building... will update if artisan builds and runs.

benwah, 2021-08-18:
Yay! Artisan runs on Apple silicon!

[image]

benwah, 2021-08-18:
cc @MAKOMO @roasterdave

MAKOMO, 2021-08-18:
Cool! Still there is a bit of a way to go for a full support or even a fat binary. I guess some modules will still fail. Eg. python-snap7/snap7 might not be available for M1 yet and needs to be compiled by hand. Wonder also about packaging. Anyhow, good to know that you came that far and that you have an M1 for testing.

MAKOMO, 2023-09-30:
Building is not a problem any longer as all libs work at least on Roasetta2 and/or are available for Apple Silicon by now.

closed 2023-09-30 by MAKOMO