artisan-roaster-scope/artisan #671: Language switching is broken

MAKOMO, 2021-08-20:
**Describe the bug**
Using the menu Config >> Language to switch to any other UI language has no effect after restart. The internal locale variable is correctly set as reflected by the menu entry, but the QTranslations are not loaded.

**Additional context**
This seems to be an negative effect of PR #659. 
@benwah

benwah, 2021-08-20:
@MAKOMO  - I've just self-assigned this. I should be able to find some time to fix it next week.

MAKOMO, 2021-08-20:
This is very tricky. Those translators have to be established at a certain time before some other classes are created. I totally forgot the exact details, but the way it was before your PR was the way it worked (with the drawback of this global var).

Thanks for looking into this.

MAKOMO, 2021-08-30:
Fixed with Commit c115619d935d5f7be196da753dffd70331e11eef

closed 2021-08-30 by MAKOMO