artisan-roaster-scope/artisan #672: Phidget 1018_2 Not Connecting

kate19058, 2021-08-24:
I am a coffee roaster out of Tumwater, Washington.  I use several different Phidget devices with my roaster and my roasting software, Artisan.  I purchased the 1018_2 Phidget to connect to my roaster and software to track the voltage.  I am able to get the Phidget to connect to my Phidget dashboard, but I am unable to get it to communicate with Artisan.  There isn't a lot of troubleshooting information available online for this Phidget with this software.  Did I purchase the incorrect device?  Or is there something that I can do to get the software and Phidget to start to talk to each other.

CarefreeBuzzBuzz, 2021-08-25:
We can’t tell if it’s correct if you don’t explain I. Detail what you are trying to do.  Please try again.  

Michael Herbert 
CarefreeBuzzBuzz

MAKOMO, 2021-08-25:
You need to use an Artisan device type of "Phidget IO 01" to eg. read channel 0 and 1 of the 1018. Note that Artisan cannot connect to the driver while the Phidget Control Panel app is open.

closed 2021-08-31 by kate19058

MAKOMO, 2021-08-31:
Would be nice to learn if and how this was resolved.