artisan-roaster-scope/artisan #676: TC4 could not convert string to float

fcebreros, 2021-08-28:
I'm using Artisan 2.4.4 on Ubuntu (couldn't run the last version )
and I get an error @Line5435 "arduinotc4temperature(): could not convert string to float"
when I start the readings

Artisan 2.4.4
Ubuntu / Linux
TC4 shield on Arduino Uno

[image]

MAKOMO, 2021-08-28:
Why couldn't you install the latest version?

fcebreros, 2021-08-28:
I could install the last version apparently without any problems but it simply didn't launch.

MAKOMO, 2021-08-28:
Could you please reinstall v2.4.6, start it via

# /usr/share/artisan/artisan

and report the error which is logged on startup to the shell such that we can investigate the issue?

fcebreros, 2021-08-28:
It seems some python stuff is missing.
I'm on Ubuntu 16.06 LTS and right now upgrading to 18.
I tried the same settings on Artisan 1.5 and got the same TC4 error.
It reads the temperature  but randomly stops because of the string to float error.
[image]

fcebreros, 2021-08-30:
Hi, 
I had the same errors trying to install the latest Artisan on Ubuntu18.
I did manage to get the TC4 board running on Artisan 2.4.4 by setting the TC4 filters to 0%

MAKOMO, 2021-08-30:
Artisan v2.4.6 is build on Ubuntu 20.

The communication error you see must be related to the TC4 firmware you are running. Best to run aArtisan_PID_6 or newer.

closed 2021-08-30 by MAKOMO

fcebreros, 2021-08-30:
OK, thanks for the info.