artisan-roaster-scope/artisan #679: HB 600 and ARTISAN :

erminiacritical, 2021-09-03:
**Describe the bug**
Good evening Marko,
this is my second help request: unfortunately the problem persists.
I have installed the latest version of Artisan 2.4.6 on my MacBook Os 11.5.2.
[IMG_9216.pdf](https://github.com/artisan-roaster-scope/artisan/files/7108579/IMG_9216.pdf)

[IMG_9215.pdf](https://github.com/artisan-roaster-scope/artisan/files/7108575/IMG_9215.pdf)

I have configured the machine
I have configured the interface on COM4, (is this correct?)
I flagged 'serial register'

However, the connection is not constant and almost always it falls on the turning point and data is lost.
MODBUS error always appears.
What can I do? Can you help me?
Thanks and good w.e.
Erminia

[IMG_9213.pdf](https://github.com/artisan-roaster-scope/artisan/files/7108573/IMG_9213.pdf)

**Setup (please complete the following information):**
 - Artisan Version [e.g. v2.4.4]: 
 - Artisan Build (number in brackets shown in the about box) [e.g. 97e6dec]: 
 - Version [e.g. 22]: 
 - Platform (Mac/Windows/Linux + OS version) [e.g. macOS 10.15.3]: 
 - Connected devices or roasting machine [e.g. Probatone 5]: 

**Additional context**
Add any other context about the problem here.

Please attach your current Artisan settings file (as exported via menu Help >> Save Settings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "No
[IMG_9213.HEIC.pdf](https://github.com/artisan-roaster-scope/artisan/files/7108569/IMG_9213.HEIC.pdf)
t a supported file type" error on uploading._

MAKOMO, 2021-09-04:
Again you did not attach your artisan-settings.aset file as requested. Your Artisan is obviously configured to communicate via MODBUS, but your roasting machine is communicating via the Arduino/TC4 protocol. Do a Factory Reset under menu Help >> Settings and reconfigure for your HB roaster using the menu Config >> Machine >> HB Standard.

closed 2021-09-04 by MAKOMO

erminiacritical, 2021-09-04:
I’ll try, tank you.

       ERMINIA NODARI
       CO-FOUNDER 
       ***@***.***
       www.critical.coffee 
       +393476093236
     
        

Le informazioni contenute nella comunicazione che precede possono essere riservate e sono, comunque, destinate esclusivamente alla persona o all'ente sopraindicati. La diffusione, distribuzione e/o copiatura del documento trasmesso da parte di qualsiasi soggetto diverso dal destinatario è proibito. Se avete ricevuto questo messaggio per errore, Vi preghiamo di contattarci immediatamente tramite e-mail e di cancellare questo messaggio.
Grazie
Confidentiality notes: the information in this message is intended to be confidential and for the use of only the individual or entity named above.
The information herein contained may considered legally privileged information. If you are not the intended recipient you are notified that retention, dissemination, distribution the than to the intended recipient, or copying of this message is strictly prohibited. If you receive this message in error please notify us. 
 Nodari
***@***.***