artisan-roaster-scope/artisan #683: 2.46 2.47  font is used for annotations within the actual profile graph display □□□

huangdaojiri, 2021-09-08:
I use the win10 operating system. I set the Language as "traditional Chinese"

In artisan version 2.44 and before, the Chinese words in the circle can be displayed normally. However, in versions 2.46 and 2.47, these places display □□□

SourceHanSansCN-Regular.otf，dijkstra.ttf,Humor-Sans.ttf,WenQuanYiZenHei-01.ttf,NanumGothic,DejaVu,Sans MonoAll fonts have been installed in win10

2.47
[image]

2.44
[image]

 I will be looking forward to your reply.

roasterdave, 2021-09-09:
Please go to Config>> Curves>> UI tab and select the font _**Source Han Sans CN**_.  Artisan has its own internal copy of these fonts so that it can run on any machine.

[image]

[image]

[image]

huangdaojiri, 2021-09-10:
Thank you so much, this problem is completely resolved.

roasterdave, 2021-09-10:
Good to hear!

closed 2021-09-10 by roasterdave