artisan-roaster-scope/artisan #685: maintenance counter

peterweiser, 2021-09-11:
Right now, I m always using tools like excel or outlook calender to schedule the maintenance intervals for cleaning the roasters parts like the exhaust pipe, the cyclone or the cooling tray, ...
It would be great to be able to tell artisan, how often these things have to be cleaned.
As it strongly correlates with the amount of operating hours and the amount of roasted coffee, I think it could work quite well.
It would be great to get a reminder or something from artisan - maybe a few days before the maintenance should be carried out. 
Further more, a protocol of the already completed maintenances would be quite helpful.
Maybe this feature is already planned to be implemented?

Greetings from Austria

Peter

CarefreeBuzzBuzz, 2021-09-11:
If it correlates to your Batches there is already a batch counter feature? 

https://artisan-scope.org/docs/batches/

There is no pop up or reminder at this point.  

___________________________
Michael Herbert, CarefreeBuzzBuzz

MAKOMO, 2022-05-02:
Maintenance reminders have been added to https://artisan.plus

closed 2022-05-02 by MAKOMO