artisan-roaster-scope/artisan #693: Version [2.4.6]

catpeer, 2021-09-19:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Setup (please complete the following information):**
 - Artisan Version [e.g. v2.4.4]: 
 - Artisan Build (number in brackets shown in the about box) [e.g. 97e6dec]: 
 - Version [e.g. 22]: 
 - Platform (Mac/Windows/Linux + OS version) [e.g. macOS 10.15.3]: 
 - Connected devices or roasting machine [e.g. Probatone 5]: 

**Additional context**
Add any other context about the problem here.

Please attach your current Artisan settings file (as exported via menu Help >> Save Settings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2021-09-19:
Hm

MAKOMO, 2021-09-19:
This is not much information and the strace is not conclusive. Which platform exactly are you trying to run Artisan on? How did you install Artisan exactly?

MAKOMO, 2021-09-19:
Note that several people are running Artisan v2.4.6 successful under Linux (mostly Ubuntu).
Maybe you also find a better title for this issue.

catpeer, 2021-09-19:
I use a Linuxmint distribution LMDE 4 which is based on Debian 10 with the kernel version 5.10.0-0.bpo.3-amd64. I dit use the version 2.1.2 up today and I installed all versions with an installation tool (GDebi). Now I installed version 2.4.4 and it runs again. The only small issue I had in the past was a long starting time.
Please excuse my akward posting, since it’s my first one github.

catpeer, 2021-09-19:
After further investigation I have to admit that the issue I do have with v2.4.6 depends on my own installation. I was able to run artisan v2.4.6 in a virtual machine.
I regret to have caused some trouble. A software considerably useful.

closed 2021-09-19 by catpeer