artisan-roaster-scope/artisan #720: Bean Temperature Can't show on screen

TonSDe, 2021-11-02:
**Describe the bug**
BT cannot be displayed on the screen. in spite of data from Modbus.

**To Reproduce**
Steps to reproduce the behavior:
1. New install Artisan 2.4.6 on macOS
2. Go to -> Config -> Machine -> Coffee-Tech -> Silon Touch
3. Setup my MacBook to IP address to 192.168.1.100
4. Click ON
5. BT show "u.u"

**Expected behavior**

**Screenshots**
Screen 1 : 
[image]

Screen 2 : Testing Scan Data from Modbus
[image]

Screen 3 : BT Can't show data
[image]

**Setup (please complete the following information):**
 - Artisan Version : 2.4.6
 - Artisan Build (number in brackets shown in the about box) [e.g. 97e6dec]:  f1b8e57
 - Version [e.g. 22]: 
 - Platform (Mac/Windows/Linux + OS version) [e.g. macOS 10.15.3]:  macOS 12.0.1
 - Connected devices or roasting machine [e.g. Probatone 5]: Silon ZR-7

**Additional context**
[silon-zr7-artisan-settings.aset.zip](https://github.com/artisan-roaster-scope/artisan/files/7457581/silon-zr7-artisan-settings.aset.zip)

MAKOMO, 2021-11-02:
Thanks for your detailed report! I did load my MODBUS simulator with the values shown in your second screenshot, started Artisan v2.4.6, loaded your Artisan settings (which look absolutely fine to me), pointed it to my simulator and Artisan showed the correctly decoded data.

[image]

I also checked if you have any custom symbolic formula for BT defined or any input filter limit set, but couldn't find neither in your settings.

Is the BT permanently not shown? Was it never shown? Is this a new machine? Did you check back with CTE?

chrisnance419, 2021-11-04:
Hello, 

The solution to this error is to install CCW software on your computer and update the software version for your roaster. These can both be supplied for free by Coffee-Tech Engineering.

closed 2021-11-06 by TonSDe