artisan-roaster-scope/artisan #732: Ability to compute and plot a temperature difference or other custom variable

aaronsmith328, 2021-11-23:
I roast on a SR800 that is instrumented with 3 thermocouples. One is in the base and reads the exhaust temp of the electric heater. The other two are inserted into the roast chamber the lower one is my bean temp and the upper one is my exit air /environment temp.

I would like to compute the difference in temperature between the exhaust temp of the heater and the bean temperature.  This difference is the driving force of my roast and dictates the RoR.  I have looked through the settings etc. I don't see a place to add a custom calculation and how to add that to my plot.

For a user experience, I imagine being able to:
1. Assign or determine a symbolic name for my current recorded temperatures.
     for this example:
     T_h - heater temp
     T_b - bean temp
     T_e - exit air temp
2. A Place to define a new variable. T_diff - my new variable I want to compute
3. A place enter a simple formula to compute the new variable. 
  T_diff = T_h - T_b

In spirit this should allow addition, subtraction, multiplication, and division.  Users may have some other formula with constants they want to input.  Users might also want to make a formula that is dependent on the roast time dependent...so being able to know the variable name for this time would be useful.

MAKOMO, 2021-11-23:
Did you check https://artisan-roasterscope.blogspot.com/2019/11/symbolic-formulas-basics-new-variables.html

closed 2021-11-23 by MAKOMO

aaronsmith328, 2021-11-23:
Thanks.  This was the documentation I could not find.before.