artisan-roaster-scope/artisan #746: △ROR

eat-peach, 2021-12-02:
Greetings.

I suggest adding an optional △ROR curve to the program. Thank you!

MAKOMO, 2021-12-02:
Why?

Did you see this: [Identifying Flicks](https://artisan-roasterscope.blogspot.com/2019/11/symbolic-formulas-basics-new-variables.html#identifyingFlicks)? This is based on △ROR.

Note the noise!

eat-peach, 2021-12-02:
Yes, I know the Identifying Flicks. But as mentioned, the noise is too loud. So I wonder if there's another way to demonstrate the Ro ROR. In other words, whether the ROR drops steadily or to project the sudden change in ROR.

MAKOMO, 2021-12-02:
How to do this reliable if the △ROR is that noisy. Note that this is consequence of the mathematical calculation which amplifies the noise originally stemming from the underlying signal even further. Every magic that can resolves this, either violates our mathematical universe or comes with other drawbacks, like extensive signal delays. There is no way around this. This is not a question of clever algorithmic.

closed 2021-12-02 by MAKOMO