artisan-roaster-scope/artisan #750: Yoctopus Virtual Hub keeps crashing eratically

HansGem, 2021-12-06:
Hopefully someone can throw light on the problem _or_ advise to change to Phidgets to solve a frustrating problem, _or_ suggest where I can get advice? 

Artisan runs magnificently - (2 years now) from two thermocouples  via yoctopus module, plus  gas pressure meter to MacBook Pro now running BigSur 112.6. 

Problem is:  erratic completely unpredictable failure  and shut down of Yoctopus virtual hub. After coach, to keep Artisan going  I simply need to click on virtual hub to restart, rand it econnects and goes well.  Can go for entire roast or suddenly fail every few minutes. 

I have replaced thermocouples etc numerous times, also checked all connections and connected in different ways.   

Here below is the VirtualHub record -  all goes well until 18:37:37 - then see 18:10:43 where crash stars. :-( 

Last login: Fri Dec  3 17:54:20 on ttys000

The default interactive shell is now zsh.
To update your account to use zsh, please run `chsh -s /bin/zsh`.
For more details, please visit https://support.apple.com/kb/HT208050.
/Users/jHansR/Desktop/VirtualHub ; exit;
MU00096573X:~ HansR$ /Users/HansR/Desktop/VirtualHub ; exit;
[2021-12-03 18:07:37] (01) Yoctopuce VirtualHub 1.10.38634 (dated Nov 29 2019)
[2021-12-03 18:07:37] (01) Serial is VIRTHUB0-a8330b7939
[2021-12-03 18:07:37] (01) NetworkName is VIRTHUB0-a8330b
[2021-12-03 18:07:37] (01) LogicalName is not set
[2021-12-03 18:07:37] (01) [1]ystream: 2286: Device METEOMK2-12DC1F plugged
[2021-12-03 18:07:37] (01) [1]ystream: 2286: Device THRMCPL1-12D6E5 plugged
[2021-12-03 18:07:37] (01) HTTP server is listening port 4444
[2021-12-03 18:07:37] (01) VXI-11 server is listening port 4445
[2021-12-03 18:10:43] (01) [1]ypkt_osx: 522: IOHIDDeviceSetReport failed with 0xe00002d8
[2021-12-03 18:10:43] (01) [1]ystream: 696: Error THRMCPL1-12D6E5(3) : IOHIDDeviceSetReport failed (ypkt_osx:523)
[2021-12-03 18:10:43] (01) 1:1:request write for dev THRMCPL1-12D6E5 error "-8:IOHIDDeviceSetReport failed (ypkt_osx:523)" (2:78)
Illegal instruction: 4
logout
Saving session...
...copying shared history...
...saving history...truncating history files...
...completed.

[Process completed]

Many thanks 

Hans R

MAKOMO, 2021-12-06:
Dear Hans, what is the reason you connect the Yocto modules via a Virtual Hub? Does the same issue occurs on direct USB connection? I have not much experience with Yocto modules in practice, only did some limited testing and never heard reports on stability problems.

Anyhow, the issue seems to be on the Yoctopuce driver/hub side and not directly related to Artisan. Did you send this log report to Yoctopuce? For them it might be able to interpret the log and advice you on how to fix this.

HansGem, 2021-12-06:
Thanks Marco: I now have a reply from Yoctopuce and looks. Like you are right - I will post any breakthrough that explains and solves the problem.

HansGem, 2021-12-08:
Dear Marco et al:
The Virtual Hub is the software Yocto supply to translate the data from the modules supplied via the USB and interpretable by Artisan. In retrospect I may have been better to use Phidgets, and may still do in the future. 
I have now run the latest version of the Yocto virtual hub (Nov 21) and the errors and 'unplugging' still occurs, however this version of the  hub does not shut down. It ignores the errors after reporting them, and reconnects the supposedly 'unplugged' thermocouples. I assume the 'unplugging' is an interruption as I have checked all wiring etc.  Since the hub nolonger shuts down, neither does Artisan!  Hurray! So all is now workable.

MAKOMO, 2021-12-08:
> Dear Marco et al: The Virtual Hub is the software Yocto supply to translate the data from the modules supplied via the USB to be interpretable by Artisan. 

I am confused by this statement. The Artisan builds contain a Yocto driver that allows to access the modules via USB directly without any further software installation. The VirtualHub is only used in "network" mode to access modules remotely eg. via WLAN which is not advisable because of the interferences caused by roasting machine motors.

> In retrospect I may have been better to use Phidgets, and may still do in the future. 

Why is this? I do not see a fundamental difference in Phidgets vs Yocto modules.

> I have now run the latest version of the Yocto virtual hub (Nov 21) and the errors and 'unplugging' still occurs, however this version of the hub does not log out and shut down. It ignores the errors after reporting them, and reconnects the supposedly 'unplugged' thermocouples. I assume the 'unplugging' is an interruption as I have checked all wiring etc. 

Most likely is a disturbance of the WIFI signal which you use to connect to the Virtual Hub!?

> Since the hub nolonger shuts down, neither does Artisan! Hurray! So all is now stable enough to be workable.

As said there is a Yocto driver installed within the Artisan build. The continous build of Artisan includes the latest Yocto driver available today. You can find those builds (macOS/Windows) here:

https://github.com/artisan-roaster-scope/artisan/releases/tag/continuous

HansGem, 2021-12-09:
Ah! all solved - I had ticked "network"  box - unticked it and it all works without the installed Virtual Hub, and the contained driver results in absolutely stable! I had latest Artisan Update. Many Tx!

closed 2021-12-13 by MAKOMO