artisan-roaster-scope/artisan #753: Request for temp label along with gas setting along timeline

fatrabbit-la, 2021-12-08:
If this is not already possible (I couldn't find it), I would like to request an option to list the BT along with changes in gas setting. The way I have things set up, my charts along the timeline show the gas setting with a line below that. Would be great to see BT below the line. It's a bit hard to tell, especially on the fly, what exactly the temp is at a given time. I use background roasts for consistency and if I make minor changes to the timing of changes, often based on BT, it's impossible to remember these detailed changes.

roasterdave, 2021-12-08:
Have you looked at [event annotations](https://artisan-roasterscope.blogspot.com/2020/05/special-events-annotations.html)?  These could give you all the details you want to see in your profile.

[image]

To see a temperature on the fly at any time, move the mouse cursor and view the time (x) and temperature (y) value which is displayed near the top.  To change the y value from the Temperature axis to the RoR axis and back, hit the 'd' key.

[image]

To toggle on or off cross hairs hit the 't' key (Ctrl+Shift+t on Windows).  

[image]

With the cross hairs on hold down the left mouse button and drag to measure differences.

[image]

Do these provide what you are looking for?

InvisibleOrdinary, 2021-12-08:
Thank you for posting that… I didn't know that existed, and I definitely see the use in that. Is it possible to lock the cross hairs and measuring to the curve (BT) so it would show deltas along the bean's trajectory - seems to me that a simple straight difference would be fine for this.

roasterdave, 2021-12-08:
No, you cannot lock the cross hairs.  However, you should be able to get want you want with a virtual device, that displays the difference in an LCD during roasting and/or as a curve.  The basics are in this [blog article](https://artisan-roasterscope.blogspot.com/2019/11/symbolic-formulas-basics-new-variables.html).  More blog articles with examples can be found with [this search](https://artisan-roasterscope.blogspot.com/search?q=virtual++devices).

InvisibleOrdinary, 2021-12-08:
Ok…  that is fine (I do have an LCD that shows what I refer to  most), but not the same thing as being able to see the delta between any two selected points. Close enough in most cases, though.

fatrabbit-la, 2021-12-08:
Yes, this is great! My only request then, because I didn’t see this as an option in the help file, would be if it would be possible to insert a [return] such that you can create a line break in order to avoid information overlapping. I do see the ability to compensate for overlap, but that ends up deleting (making invisible) data.

Many thanks for pointing this out.

Dan

Fat Rabbit Coffee
***@***.*** ***@***.***>
www.fatrabbitcoffee.com <http://www.fatrabbitcoffee.com/>

roasterdave, 2021-12-09:
Dan, there is no way to insert a line break at this time.  One other thing to try is to increase the size of the steps so the annotations have some separation.  Change the setting for Config>> Events>> 100% Event Step.  Increasing the value will stretch the steps upward.

fatrabbit-la, 2021-12-10:
David,

Can you please clarify? I’m on Config > Events and I see Step as an option for Markers (that’s what I currently use) but I don’t see where to set Event Step to 100%. To the far right of that I see values I can adjust from 1 - 20. It doesn’t seem to make a difference but perhaps that’s just because of the particular graph I’m looking at.

I looked on other tabs to adjust Event Step but only see it on the Config page.

Dan

Fat Rabbit Coffee
***@***.*** ***@***.***>
www.fatrabbitcoffee.com <http://www.fatrabbitcoffee.com/>

CarefreeBuzzBuzz, 2021-12-10:
https://artisan-roasterscope.blogspot.com/2020/05/special-events-annotations.html <https://artisan-roasterscope.blogspot.com/2020/05/special-events-annotations.html>

Read this entire blog post for clarification on items.  There are often blog posts on the various features.  

That’s the first place to go.  Hope this helps.  

----------------------------------------
Michael Herbert, Head Roaster
Order at www.carefreebuzzbuzz.com <http://www.carefreebuzzbuzz.com/>
Artisan RoasterScope  <https://artisan-scope.org/>Team

roasterdave, 2021-12-10:
Dan, the correct location is Config>> Axes.  Sorry for the confusion.

closed 2021-12-13 by MAKOMO