artisan-roaster-scope/artisan #755: ROR graph, dry%, ramp%, >>fc all not displaying

ericschlo, 2021-12-13:
**Describe the bug**
Hi, I have been having the issue where my ROR line isn't showing, nor is the turning point temp/time, and everything in the display panels on top. I will restart the program, and usually my first roast after restarting charts correctly, but the second or third roast in a row will not display these things. Sometimes even after restarting the program it doesn't display them, and it's super frustrating because I don't find out whether or not it's going to work until I’m mid roast. How do I fix this error?

**Expected behavior**
I launch artisan, plug in my 301 control to usb, warmup my machine, press on, start, then charge. I begin my roast. everything works correctly. I drop, save as, and reset. press on. start. charge. then, just after turning point, there is no marker for turning point noting time and temp. now I know it's not going to log my ROR or any vitals. the BT and ET still are logged, but nothing else. v frustrating. must finish roast without seeing the ror graph and restart program.

**Screenshots**
attached -- when this problem occurs, the dry% and ramp% and dev time give some wild numbers.

**Setup (please complete the following information):**
Artisan 2.4.6 build 2 (f1b8e57) 
Python 3.9.1, Qt 5.15.2, PyQt 5.15.4, Matplotlib 3.4.2, NumPy 1.21.1, SciPy 1.7.0, pymodbus 2.5.2, Yoctopuce 1.10.45664 (1.10.45664) 
I’m using MAC OS Monterey 12.0.1
connected to huky 500 via USB center 301 type k computer.

[image]

[github.zip](https://github.com/artisan-roaster-scope/artisan/files/7700395/github.zip)

CarefreeBuzzBuzz, 2021-12-13:
Did you ever push the charge button.  Look in Roast Properties > data to
see.

MAKOMO, 2021-12-13:
Thanks for the detailed report. Michel is right. The CHARGE event is missing. This is an FAQ. Please see "[I can not see the rate-of-rise curve during roasting?](https://artisan-scope.org/help/#FAQ-Recording-7)".

closed 2021-12-13 by MAKOMO