artisan-roaster-scope/artisan #758: ERROR MODBUS AND SERIAL EXCEPTION- MACPRO VS HB600

erminiacritical, 2021-12-15:
Hi Marko,
I am very happy because I think I have solved the problem that bars Artisan and HB600 to run together.
I share in the blog:
To avoid:
  - Modbus 1 communication error 14: 23: 09.740 Modbus error: writeMask () Modbus Error: [Connection] Failed to connect [ModbusSerialClient (rtu baud [115200])] @ line 493
- Serial Exception: Not able to open serial port
After installing the driver CP210X VPC on the Mac (remember to open the privacy lock)
Connect Artisan with the roaster turned on and in the Port Configuration screen choose CP2102 USB to UART Bridge Controller
In the Modbus screen: choose v / cu.usbserial-0001.
Thanks for the cooperation Marko !!!!! I was born in the pre-digital era and I am really satisfied when I overcome the obstacle :)
Erminia

[image]
[image]
[image]
[image]

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Setup (please complete the following information):**
 - Artisan Version [e.g. v2.4.4]: 
 - Artisan Build (number in brackets shown in the about box) [e.g. 97e6dec]: 
 - Version [e.g. 22]: 
 - Platform (Mac/Windows/Linux + OS version) [e.g. macOS 10.15.3]: 
 - Connected devices or roasting machine [e.g. Probatone 5]: 

[image]

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2021-12-16:
Thanks for this confirmation!

closed 2021-12-16 by MAKOMO

erminiacritical, 2021-12-17:
Hi, I still have problems. This is just a test to see if I can reopen this thread just by commenting.

erminiacritical, 2021-12-17:
I’m @erminiacritical
👀 Hi Marko, my problem is not solved.( HB600)  I roasted yesterday only once seeing the ET / BT values on the display even in the presence of the MODBUS error. Today I don't see the values and therefore I have to roast without curve and without RoR. I attach a list of errors and screenshots. Where should I change? What should I change? Please help me
[17dicembre.docx](https://github.com/artisan-roaster-scope/artisan/files/7736637/17dicembre.docx)
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

MAKOMO, 2021-12-17:
Hi,

your setup is (again) totally broken. Your machine is NOT communicating via MODBUS thus your should not have MODBUS configured at all.

I can only guess that you loaded an .alog profile recorded on a different setup (one with MODBUS configured) and answered the dialog

[image]

with YES instead of NO. As a consequence your setup got modified.

Do the following:

1) Factory Reset
  menu Help >> Factory Reset
2) Setup again for your HB roaster
  menu Config >> Machine >> HB >> ..
as you did before.