artisan-roaster-scope/artisan #781: Trouble with 2.4.6 on Win 11

dopeslope, 2022-01-04:
**Describe the bug**
entire program looses responsiveness across buttons after roughly 15 seconds, and or after selecting a menu item or two. I get a null beep when every I try to engage any button or menu.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Setup (please complete the following information):**
 - Artisan Version 2.4.6
 - Artisan Build f1b8e57
 - Version [e.g. 22]:  Platform (Windows 11 Home v. 21h2) 
 - Connected devices or roasting machine [Hottop kn-8828b-2k ] 

**Additional context**
worked just fine for two previous roasts. This last round I noticed that RoR was not tracking, then dropped beans...and was unable to end the cooling cycle or terminate the roast in Artisan. it just kept running and tracking basic stats until i forced it to shutdown. upon restart it began stalling. I reinstalled and it is doing the same. 

Please attach your current Artisan settings file (as exported via menu Help >> Save Settings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._
[artisan-settings.txt.zip](https://github.com/artisan-roaster-scope/artisan/files/7809388/artisan-settings.txt.zip)

MAKOMO, 2022-01-05:
Windows 11 was released after the current Artisan release. We have no experience running Artisan on Windows 11 and for now we have to declare it unsupported on this platform until we get access to a corresponding system in the future, maybe for the but next version.

MAKOMO, 2022-02-27:
We upgraded the GUI lib underlying Artisan. That new version should support Windows 11. Could you please confirm this by running the continuous builds?

https://github.com/artisan-roaster-scope/artisan/releases/tag/continuous

MAKOMO, 2022-03-04:
Did you had a chance to test this?

dopeslope, 2022-03-04:
Hello,

I will try this early next week.

MAKOMO, 2022-03-14:
This is sounds like a bug in the Qt GUI lib underlying Artisan. We can only hope that this get fixed some day. Nothing we can do about here so I close this one.

closed 2022-03-14 by MAKOMO