artisan-roaster-scope/artisan #786: Add SYMBOLIC VARIABLES for Ambient Sensors (Temperature, Humidity, Pressure)

bigl00z3, 2022-01-12:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

I wanted to calculate the temperature increase of my motor from starting temperature (Ambient). 
Background: My roaster is in my Garden shed which means that I get high temperature variations. Up to +35°C in the summer when in the sun to -10°C in the winter. Tonight it's already at -2°C out there. still debating if I go fout to roast one batch or not ;-) 
In the Summer I had one time a thermal switch stop due to a motor overheat and in the winter below a certain temperature it is a bit squeaky until it's "warmish".  This is why I initially added a thermocouple on my motor.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

SYMBOLIC VARIABLES are already existing for most values. But I haven't found it for the ambient sensors. They may already exist but are not documented in _Config/Device.../Extra Devices/Help_

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

I thought about adding extra devices, but that would eat some port space on my Phidget HUB. 
And they are already devices connected that could do the job. In my case Phidget HUM1000 and PRE1000.

**Additional context**
Add any other context or screenshots about the feature request here.
-

MAKOMO, 2022-01-12:
Currently the ambient sensors are harvested only once short after turning Artisan ON. That value could be bound to a symbolic variable, however, this value would be constant during the roast until Artisan is again turned on. These ambient values could easily bound to symbolic variables.

In case you plan to harvest the ambient devices continously during a roast one would need to add a corresponding extra device type for those ambient devices. However, those sensors are not really made for this type of application.

bigl00z3, 2022-01-12:
I would imagine that 1 static value taken once per roast (at start) would be more than sufficient. 
If continuous data would be required I would agree that it would be better to use other sensors directly mapped in "Extra Devices"

MAKOMO, 2022-01-13:
Added the symbolic variables aTMP, aHUM and aPRE which hold the last ambient temperature, humidity and pressure readings (Commit 73f346b497ad3600535a015547a15410a11368c8)

closed 2022-01-13 by MAKOMO