artisan-roaster-scope/artisan #797: Unable to open serial port on MACOS Monterey

yoln77, 2022-01-26:
**Description of the problem**
Hi Marko, I receive an 'unable to open serial port' error once connecting a center 306 to a macookpro on MacOS Monterey. 

I Followed your advices on request #456 (https://github.com/artisan-roaster-scope/artisan/issues/456) before posting this issue.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Config'->'Port...' tab 'ET/BT' (the 'Extra' Tab is absolutely empty, I can't click on anything there contrary to what is mentioned on the above link #456)
2. Select 'CP2102 USB to UART Bridge Controller' which is pre-populated in the drop-down menu
3. See that serial port setting is now : /dev/cu.usbserial0001,9600,8,N,1,1.0
4. Click 'ON'
5. See the error  'Unable to open serial port'

**Expected behavior**
Reading of the temperature

**Screenshots**

[image]
[image]

**Setup :**
 - Artisan Version  v2.4.6 build 2
 - Artisan Build f1b8e57
 - Platform: macOS 12.0.1: 
 - Connected devices : Center 306 

**Additional context**

When running ls /dev/tty.* I see   /dev/tty.usbserial-0001   which seems identical to the one auto-populated by artisan

Also here is what I see in my USB port settings: 

[image]

Please attach your current Artisan settings file (as exported via menu Help >> Save Settings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.
[artisan-settings.txt](https://github.com/artisan-roaster-scope/artisan/files/7938802/artisan-settings.txt)

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2022-01-26:
Are you running on an M1 processor? Do you connect the Center with an original USB-C to USB adapter from Apple?

yoln77, 2022-01-27:
Hi, updating the driver did the trick!

And yes, Original cabled M1-Pro chip

Thanks a lot for your help!

closed 2022-01-27 by yoln77

MAKOMO, 2022-01-27:
Thanks for this feedback. Good to know that it works even on M1 with the updated driver.