artisan-roaster-scope/artisan #805: Broken IO Command "set(c,0[,sn])" (off) in continuous build.  "set(c,1[,sn])" (on) still works.

bigl00z3, 2022-01-30:
Continuous build:  2ae042e and  5ebb838 seem to have broken the  IO Command `set(c,0[,sn]) (0=off)`

Command ON is working: `set(7,1,595251)`
Command OFF is **not working** anymore: `set(7,0,595251)` 

The command is working fine in previous 2.4.7 versions. Tested successfully in df1b13a

Issue (and previous working state) has been tested both in OS X and Windows.

Nico

MAKOMO, 2022-01-30:
Thanks for reporting. Tried to improve things and broke them. Should be fixed with the next commit. You should then also be able to use ```set(7,true,595251)``` and ```set(7,False,595251)```.

closed 2022-01-30 by MAKOMO

MAKOMO, 2022-02-01:
Commit is in with many other fixes and improvements..