artisan-roaster-scope/artisan #812: Artisan + Phidgets + Ununtu Linux ..Not working

MAKOMO, 2022-02-05:
### Discussed in https://github.com/artisan-roaster-scope/artisan/discussions/811

<div type='discussions-op-text'>

<sup>Originally posted by **balliet** February  5, 2022</sup>
I recently purchaed VINT HUB0001_0  and TMP1101_1 and 2 thermocouples. 

I am using Ubuntu 20.04

I installed all of the hardware according to the instrctions and passed the HelloWorld test to make sure everything was working. 

I changed the 99-libphidget22.rules file to the following:
SUBSYSTEMS=="spi", DRIVERS=="phidget_hub0001_0", ACTION=="add", MODE="666"

After entering Artisan and changing the device and machine setting to use the pfidgets, I thought I was ready to go.

clicked the default ON button.  it display showed that "SCOPE HAS BEEN RESET"  BT and ET was displayed  but there was no  temperature reading displayed.  

I have been working on this for several days and have reloaded the system anew, and reinstalled the Phidgets software again and I get the same results

Need Help .... Thank you 

</div>

MAKOMO, 2022-02-05:
On my Mac with the latest Phidgets driver installed Artisan v2.4.4 and v2.4.6 work with the original HUB0000 with the latest and but latest firmware version as well as with the HUB0001 with the latest (only available) firmware version. Thus there seems to be no issue on on macOS.

However, the latest HUB0000 firmware (307) and the HUB0001 111 firmware seems not to be supported by Artisan v2.4.6 on Linux and RPi. Windows is untested.

MAKOMO, 2022-02-07:
The root cause for those troubles are that the Linux build of Artisan v2.4.6 contained a bundled Phidget driver which now is outdated and not supporting the HUB0001. Installing a newer system-wide Phidget driver did not help as Artisan prefers the bundled one.

The Phidget driver is no longer bundled with Linux builds (as it never was with macOS or Windows builds) to allow user to update their drivers.

closed 2022-02-07 by MAKOMO

wvengen, 2024-08-23:
Note that the Flatpak version of Artisan (#1602) does bundle the Phidget driver (necessarily).
The Phidget driver source is [being monitored for updates](https://github.com/flathub/org.artisan_scope.artisan/blob/9d5f356d7833549d99de9a60446c437587c1bb9e/org.artisan_scope.artisan.yml#L56-L65), so when a new Phidget driver is released, the Flatpak is meant to get an update easily.