artisan-roaster-scope/artisan #820: Error Message: Exception: sample() The truth value of an array with more than one element is ambiguous.

bigl00z3, 2022-02-15:
In current 2.4.7. Continuous build 00d8488 there seem to be something that has changed. 
I keep getting the following error when pas CHARGE:

`91 08:58:21.618 Exception: sample() The truth value of an array with more than one element is ambiguous. Use a.any() or a.all()@line 4171`

In previous continuous 2.4.7 build b196b64 It runs fine without error

Nico

bigl00z3, 2022-02-15:
Here is an extract of the log file:

`02-15 08:53:43 Dummy-18   artisanlib.s7port    DEBUG  : writeInt(5,2,14,93)
02-15 08:53:43 MainThread artisanlib.main      DEBUG  : message: Exception: sample() The truth value of an array with more than one element is ambiguous. Use a.any() or a.all()@line 4171
02-15 08:53:44 Dummy-12   artisanlib.s7port    DEBUG  : readActiveRegisters()
02-15 08:53:44 Dummy-12   artisanlib.s7port    DEBUG  : clearReadingsCache()
02-15 08:53:44 Dummy-12   artisanlib.s7port    DEBUG  : readInt(5,2,38,False)
02-15 08:53:44 Dummy-12   artisanlib.s7port    DEBUG  : return cached value => 184
02-15 08:53:44 Dummy-12   artisanlib.s7port    DEBUG  : readInt(5,2,36,False)
02-15 08:53:44 Dummy-12   artisanlib.s7port    DEBUG  : return cached value => 206
02-15 08:53:44 Dummy-12   artisanlib.s7port    DEBUG  : readInt(5,2,10,False)
02-15 08:53:44 Dummy-12   artisanlib.s7port    DEBUG  : return cached value => 250
02-15 08:53:44 Dummy-12   artisanlib.s7port    DEBUG  : readInt(5,2,14,False)
02-15 08:53:44 Dummy-12   artisanlib.s7port    DEBUG  : return cached value => 93
02-15 08:53:44 Dummy-12   artisanlib.s7port    DEBUG  : readInt(5,2,48,False)
02-15 08:53:44 Dummy-12   artisanlib.s7port    DEBUG  : return cached value => 92
02-15 08:53:44 Dummy-12   artisanlib.s7port    DEBUG  : readInt(5,2,16,False)
02-15 08:53:44 Dummy-12   artisanlib.s7port    DEBUG  : return cached value => 110
02-15 08:53:44 Dummy-12   artisanlib.s7port    DEBUG  : readInt(5,2,52,False)
02-15 08:53:44 Dummy-12   artisanlib.s7port    DEBUG  : return cached value => 110
02-15 08:53:44 Dummy-12   artisanlib.s7port    DEBUG  : readInt(5,2,18,False)
02-15 08:53:44 Dummy-12   artisanlib.s7port    DEBUG  : return cached value => 72
02-15 08:53:44 Dummy-12   artisanlib.phidgets  DEBUG  : getFirstMatchingPhidget(PhidgetTemperatureSensor,23,None,False,False,None,None)
02-15 08:53:44 Dummy-12   artisanlib.phidgets  DEBUG  : getFirstMatchingPhidget(PhidgetTemperatureSensor,23,None,False,False,None,None)
02-15 08:53:44 MainThread artisanlib.main      ERROR  : The truth value of an array with more than one element is ambiguous. Use a.any() or a.all()
Traceback (most recent call last):
  File "/private/var/folders/pp/pstbdyms02533hsds8mqkk440000gn/T/AppTranslocation/4FE2CC61-0A3F-4C5B-857D-5602C1AED9A9/d/Artisan.app/Contents/Resources/lib/python3.10/artisanlib/main.py", line 4171, in sample_processing
  File "/private/var/folders/pp/pstbdyms02533hsds8mqkk440000gn/T/AppTranslocation/4FE2CC61-0A3F-4C5B-857D-5602C1AED9A9/d/Artisan.app/Contents/Resources/lib/python3.10/artisanlib/pid_control.py", line 1515, in calcSV
  File "/private/var/folders/pp/pstbdyms02533hsds8mqkk440000gn/T/AppTranslocation/4FE2CC61-0A3F-4C5B-857D-5602C1AED9A9/d/Artisan.app/Contents/Resources/lib/python3.10/artisanlib/main.py", line 14393, in backgroundSmoothedBTat
  File "/private/var/folders/pp/pstbdyms02533hsds8mqkk440000gn/T/AppTranslocation/4FE2CC61-0A3F-4C5B-857D-5602C1AED9A9/d/Artisan.app/Contents/Resources/lib/python3.10/artisanlib/main.py", line 14339, in timetemparray2temp
ValueError: The truth value of an array with more than one element is ambiguous. Use a.any() or a.all()
02-15 08:53:44 Dummy-18   artisanlib.main      ERROR  : The truth value of an array with more than one element is ambiguous. Use a.any() or a.all()
Traceback (most recent call last):
  File "/private/var/folders/pp/pstbdyms02533hsds8mqkk440000gn/T/AppTranslocation/4FE2CC61-0A3F-4C5B-857D-5602C1AED9A9/d/Artisan.app/Contents/Resources/lib/python3.10/artisanlib/main.py", line 22784, in eventaction_internal
  File "/private/var/folders/pp/pstbdyms02533hsds8mqkk440000gn/T/AppTranslocation/4FE2CC61-0A3F-4C5B-857D-5602C1AED9A9/d/Artisan.app/Contents/Resources/lib/python3.10/artisanlib/main.py", line 14393, in backgroundSmoothedBTat
  File "/private/var/folders/pp/pstbdyms02533hsds8mqkk440000gn/T/AppTranslocation/4FE2CC61-0A3F-4C5B-857D-5602C1AED9A9/d/Artisan.app/Contents/Resources/lib/python3.10/artisanlib/main.py", line 14339, in timetemparray2temp
ValueError: The truth value of an array with more than one element is ambiguous. Use a.any() or a.all()
02-15 08:53:44 Dummy-18   artisanlib.s7port    DEBUG  : writeInt(5,2,14,99)
02-15 08:53:44 MainThread artisanlib.main      DEBUG  : message: Exception: sample() The truth value of an array with more than one element is ambiguous. Use a.any() or a.all()@line 4171
02-15 08:53:45 Dummy-12   artisanlib.s7port    DEBUG  : readActiveRegisters()
02-15 08:53:45 Dummy-12   artisanlib.s7port    DEBUG  : clearReadingsCache()
02-15 08:53:45 Dummy-12   artisanlib.s7port    DEBUG  : readInt(5,2,38,False)
02-15 08:53:45 Dummy-12   artisanlib.s7port    DEBUG  : return cached value => 184
02-15 08:53:45 Dummy-12   artisanlib.s7port    DEBUG  : readInt(5,2,36,False)
02-15 08:53:45 Dummy-12   artisanlib.s7port    DEBUG  : return cached value => 206
02-15 08:53:45 Dummy-12   artisanlib.s7port    DEBUG  : readInt(5,2,10,False)
02-15 08:53:45 Dummy-12   artisanlib.s7port    DEBUG  : return cached value => 250
02-15 08:53:45 Dummy-12   artisanlib.s7port    DEBUG  : readInt(5,2,14,False)
02-15 08:53:45 Dummy-12   artisanlib.s7port    DEBUG  : return cached value => 99
02-15 08:53:45 Dummy-12   artisanlib.s7port    DEBUG  : readInt(5,2,48,False)
02-15 08:53:45 Dummy-12   artisanlib.s7port    DEBUG  : return cached value => 98
02-15 08:53:45 Dummy-12   artisanlib.s7port    DEBUG  : readInt(5,2,16,False)
[... cut ...]
  File "/private/var/folders/pp/pstbdyms02533hsds8mqkk440000gn/T/AppTranslocation/4FE2CC61-0A3F-4C5B-857D-5602C1AED9A9/d/Artisan.app/Contents/Resources/lib/python3.10/artisanlib/main.py", line 14339, in timetemparray2temp
ValueError: The truth value of an array with more than one element is ambiguous. Use a.any() or a.all()
02-15 08:53:45 MainThread artisanlib.main      DEBUG  : message: Exception: sample() The truth value of an array with more than one element is ambiguous. Use a.any() or a.all()@line 4171`

MAKOMO, 2022-02-15:
Thanks for the report. Should be fixed by Commit c0de96e4f05822bd7652b302f58f9527956232a1. Builds should be online in about 30min. Note that Continuous builds are usually unstable, sometimes not even startup at all and in general are untested. Especially in development phases were fundamental changes to the core are applied as currently. Sorry for this.

closed 2022-02-15 by MAKOMO

bigl00z3, 2022-02-15:
No problem. That's why I test them. Full version will be even more stable 😜