artisan-roaster-scope/artisan #831: Provide smaller package by specifying dependencies

edgimar, 2022-03-02:
The release packages currently bundle together all libraries that artisan depends on.  Please consider additionally creating smaller packages that specify package dependencies in order to provide the needed libraries.

edgimar, 2022-03-08:
Why not? :)  It's a security risk to have such a large number of libraries bundled with the application (someone could easily swap out a library for a malicious library), and having it depend on packages of a linux distribution opens the way for allowing it to be distributed as part of the packaging system, or at least via something like pip -- both would make it easier to upgrade, and the former would offer better security (e.g. signed packages, security audits, etc.).  If your concern is with making it generally accessible for all linux distributions, then you could use something like flatpak or snap.  Does that make sense?

MAKOMO, 2022-03-08:
The windows builds we distribute are all signed and the macOS builds are signed and notarized. Without signing some features do not work under macOS for lack of access to certain APIs, same might happen soon on Windows. I am not too much into Linux, but the Artisan build process is complex due to the different nature of packages used. Not all are available on pip, some need compilation from source. Not sure how to distribute this in an "unbundled" way such that a user ends up with a clickable app. It might be just my lack of knowledge and talent on the Linux side that hinders an improvement here. Could you make a suggestion?

mrpenner, 2022-04-07:
I think it makes more sense for someone familiar with a particular Linux distro to do the packaging. I use Debian, so I am more familiar with its packaging than I am with other distros. Debian packages are often packaged by a Debian contributor, not the upstream developer. I wouldn't expect a developer to become familiar with all Linux distros' packaging systems. There are Artisan AppImages available already, those should run on most distros.

edgimar, 2022-04-10:
I agree with the idea that distro-specific packaging should be handled by
people familiar with the specific packaging system.  Perhaps in this case
what would be ideal is to publish Artisan on PyPI, specifying all
dependencies so that it can be installed via pip (without needing to bundle
libraries as part of the distribution). If it can be installed with pip,
then this should make it easier for people to create distro-specific
packages.

I have created a requirements file that seems to work for Linux, if that
would be helpful.

MAKOMO, 2022-04-10:
Wonder how you manage the installation of the non-pip binary packages like snap in a PyPi. Could you please open a PR with your changes to create and establish a PyPi?

edgimar, 2022-04-11:
@MAKOMO here's an example that is working for linux in a venv:  https://github.com/edgimar/artisan/commit/da138370bad6e2f32d5d727112c2c7e475875285

The following works for me:
1. create and activate a python virtual env
2. install the requirements (`pip3 install -r requirments.txt`)
3. build the wheel (`pip3 wheel .` from within src folder)
4. install the wheel (`pip3 install name_of_file.whl`)
5. run artisan (assumes that bin folder is in current path)

If this all works, then it can be tested for Windows and OSX also, and the requirements.txt file adusted as needed to provide platform specific details.  Currently the `setup.cfg` file contains some dummy metadata that should be updated.  If it is working on all three platforms, then it shouldn't be too hard to push this to PyPI (some info on how to do this is [here](https://packaging.python.org/en/latest/tutorials/packaging-projects/)).

MAKOMO, 2024-06-08:
Closing this as not planned

closed 2024-06-08 by MAKOMO