artisan-roaster-scope/artisan #834: WIN 11:  When  Cursor Touches graph: TEMP/TIME not same as past

SAS613, 2022-03-05:
**Describe the bug**
.
When touching cursor to any line on graph, the temp and time at the top of screen is not bold, in a different location and different format.

[image]

MAKOMO, 2022-03-05:
Thanks for your report! Could you please clarify what exactly you mean by "touching cursor to any line on graph"? Are you running Artisan on Windows 11? Could you please send us your Artisan log file to investigate this one. For this please hold the ALT key and click the plus icon in the upper left corner. This should open an email with the log file attached. Send this please to info@artisan.plus. Thank you very much for your support!

SAS613, 2022-03-07:
Maybe I posted too soon.  Older versions just display x=2:40  y=300.  I think Time- 2:40; Temp=300 is better.  I think it should be in bold though.
Moshe

From: Marko Luther
Sent: Friday, March 4, 2022 11:21 PM
To: artisan-roaster-scope/artisan
Cc: SAS613; Author
Subject: Re: [artisan-roaster-scope/artisan] WIN 11: When Cursor Touchesgraph: TEMP/TIME not same as past (Issue #834)

Thanks for your report! Could you please clarify what exactly you mean by "touching cursor to any line on graph"? Are you running Artisan on Windows 11? Could you please send us your Artisan log file to investigate this one. For this please hold the ALT key and click the plus icon in the upper left corner. This should open an email with the log file attached. Send this please to ***@***.*** Thank you very much for your support!

closed 2022-05-02 by MAKOMO