artisan-roaster-scope/artisan #842: Tappecue WiFi Thermometer integration

davidfchuck, 2022-03-11:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
I'm new to coffee roasting. I'm building a custom roaster with PID temp controls etc.
I have a Tappecue WiFi enabled BBQ thermometer that has 4 probe plugs for control of up to 8 sensors.  
I reached out to tappecue and they make sessions public here:  https://frankdice.github.io/tappecue-api

Also, the WIfi unit can be setup in Offline mode, which then can transmit it's own SSID and temps via UDP.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

I'd love integration of the public API into Artisan... for easy click operation.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

I've just placed an order for necessary Phidgets to setup BT and ET temp feed into Artisan... it was $110.
I'd save the $ and use my Tappecue if possible.  Not sure how many other Tappecue owners there are, but it's worth asking...

**Additional context**
Add any other context or screenshots about the feature request here.

The toaster also is a "smart device" and has the same Esspressif wifi card that my Tappecue has in it.  Has there been any work done to integrate with AIoT devices / infrustructure?  VeSync is the app that communicates / controls the oven via wifi.

davidfchuck, 2022-03-11:
A side note.... the public API address https://frankdice.github.io/tappecue-api   has instructions on how to program simple API to pull temp data.  I'm not a coder, however.
I tried to SSL into the site, without success.

roasterdave, 2022-03-15:
Artisan internally supports a wide range of devices commonly used with roasting machines.  For other devices, Artisan can call a user provided external program that reads from the device and presents the data to Artisan.  An external program is how you will need to interface this device to Artisan.  

Coffee roasting is far more demanding than what this device was designed for.  The manufacturer provides no specifications for response time or update rate.  However, the following note was copied directly from the Probe Temperature call in the public API document linked from your post. "Please note that due to caching this temperature will only update every 2 minutes."  Clearly not appropriate for coffee roasting.

closed 2022-03-15 by roasterdave