artisan-roaster-scope/artisan #843: Et & Extra device temp is showing,  Bt not showing in the new version. Macos

TulumCoffee, 2022-03-12:
Hi guys. I need some help connecting the artisan v 2.4.6 to my MacBook pro. Earlier we were using a window laptop with V0.9.9 (very old) 

In this new laptop, with the same port configurations settings as the windows, nothing seemed to happen, but when I played around with the numbers, The ET & the Extra device ( Drum ) temperature is visible. The Bean temp data is missing. 

[image]

[image]

**Setup (please complete the following information):**
 - Artisan Version -2.4.6 
 - Artisan Build - f1b8e57
 - Version [e.g. 22]: I am not sure.
 - Platform - macOS Monterrey 12.2
 - Connected devices or roasting machine - Modbus
 -

Please attach your current Artisan settings file (as exported via menu Help >> Save Settings as *.aset) file.
[artisan-settings.aset.zip](https://github.com/artisan-roaster-scope/artisan/files/8237114/artisan-settings.aset.zip) 

Thank you, 
Pallav Haria
Mumbai, India

TulumCoffee, 2022-03-12:
Also, this is the Data recording hardware we have on our roasting machine
[image]
.

bigl00z3, 2022-03-12:
Probably not an [Artisan issue](https://github.com/artisan-roaster-scope/artisan/issues) it looks more like a configuration issue on your side, this probably should be closed and you should instead post this in the [discussion page](https://github.com/artisan-roaster-scope/artisan/discussions).

You should probably also install the latest [2.4.6 version](https://github.com/artisan-roaster-scope/artisan/releases/tag/v2.6.0) that just came out.

Now regarding your issue:

- You have a "Modbus communication Error" (top left of your screenshot), something wrong in your config.
- On the Mac, did you re create the configuration? Or did you import settings from the Windows config?
- Are you sure about the values in "Input 2"?
- On the Windows machine do you have any other values in `Config/Device/Extra Devices` or `Config/Device/Symb ET/BT`?
- Do you still have the Artisan settings file form the Windows machine? Maybe you could also upload for comparison.
- I imagine you did not reconnect any wires or hardware changes on the roaster itself when you changed computer?

Nico

TulumCoffee, 2022-03-12:
Hey Nico, 

- On the mac, I re created the configuration 
- About the values - I did the scanning - And under the fct3 The values were 2(3), 4(3) 6(3)
- On the windows machine- I have an Extra device as MODBUS 34 - I am not able to configure them further ( Com port, etc)
- I'll try to get the windows settings as well. 
- No, I haven't touched the wiring or made any hardware changes
- I also upgraded to the version 2.6 yesterday, but it wasn't seeming to function properly, so I downgraded back to 2.4.6

bigl00z3, 2022-03-12:
You should use the latest version. Don't downgrade, it won't fix things and it won't be supported.

On the Mac, make sure that if you use a USB interface that the adapter is not the cause (see the mention by Marko on [this post](https://github.com/artisan-roaster-scope/artisan/discussions/840))

If you get your settings from the Windows machine and import them into Artisan on your Mac, the only thing you should have to do are:

- Install the driver for your MODBUS interface. Make sure there is no Security messages regarding the drive in `SystemPreferences/Security & Privacy/General` (before opening Artisan, unplug/replug the device after installing, and also eventually a reboot) 
- Select the correct `Comm port` (try connecting/disconnecting your device if you are not sure which one it is) for your device in the Modbus tab in `Config/Port...`

Once you do this, it should work as before

Nico

TulumCoffee, 2022-03-14:
I have installed a driver, now Modbus error is fixed. but still the BT Temp isnt showing

I also loaded the settings from windows , attached below, and changed the comm port but that didnt work either.

[image]

Settings -
[artisan-settings.zip](https://github.com/artisan-roaster-scope/artisan/files/8243305/artisan-settings.zip)

bigl00z3, 2022-03-18:
Looking at your settings, I didn not see anything that seemed wrong.
Just to make sure, when you are using those settings on the Windows machine BT and ET are both working?

The only things I could think of when looking at the settings would be (at least that's what I would check if I did not get any data): 

- Is the BT Probe still well connected, is it reading anything (you should test with the Windows computer to check if it still is giving values)
- The Modbus Register for BT is not "1" but an other value (but that would not make any sense if it was working on the Window  computer with "1")
- Have you tried the other possibilities for "Input 1". `Function`  is currently on 3, what about if you try 1, 2 or 4? and `Decode` I would expect `Float` as it is for the others, but maybe this one is not (I would doubt that, but you can always try)

But I am running out of ideas.

Nico

MAKOMO, 2022-05-02:
Stall. I close this here as it seems not to be a bug in the software but a machine specific configuration problem for an unidentified roasting machine.

closed 2022-05-02 by MAKOMO