artisan-roaster-scope/artisan #847: Ramp/Soak not working correctly in v2.60?

Brewzologist, 2022-03-14:
Hi.  Upgraded from v2.4.6 today to v2.6.0 using existing aset file.  I have warmup and BBP routines I use with the software PID ramp/soak feature and they've been working well for a long time.  Today after upgrading I began warming up my roaster and noticed the BT LCD was displaying a temp that was about 40-50 degrees F lower than the current step target for all steps.  This happened with both my warmup and BBP routines.  (I noted the ET LCD temperature and andecdotally it seemed the PID was tracking to ET and not BT, even though my PID routines use the BT. ) Since time was short, I rolled back to v2.4.6 and went back to roasting with no issues.

**To Reproduce**
Steps to reproduce the behavior:
1. In v2.6.0, Launch PID dialog and start either the Warmup or BBP ramp/soak
2. Note BT temp LCD is consistently ~50 degrees F lower than the current step/target for all steps.
3. Note ET temp LCD appears to be tracking to the PID target (but not proven).

**Expected behavior**
v2.6.0 software ramp/soak should work the same as in v2.4.6

**Screenshots**
Apologies but no time to capture. Busy roasting day and didn't have time.  No other logs available as I aborted the roast and rolled back my software.

**Setup (please complete the following information):**
 - Artisan Version [e.g. v2.4.4]: v2.6.0
 - Artisan Build (number in brackets shown in the about box) [e.g. 97e6dec]: 
 - Version [e.g. 22]: 
 - Platform (Mac/Windows/Linux + OS version) [e.g. macOS 10.15.3]: Windows 10 with latest updates
 - Connected devices or roasting machine [e.g. Probatone 5]: Huky 500T using Arduino and servo controller for gas and PWM fan.  

**Additional context**
Sorry, didn't have time to really debug this due to needing to get roasts done.  Will try to reproduce but wanted to at least get you a preliminary bug report in case it's real.  Let me know if there's any specific info you want me to capture.

Please attach your current Artisan settings file (as exported via menu Help >> Save Settings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._
[artisan-settings-stevek-v4.0b.aset.txt](https://github.com/artisan-roaster-scope/artisan/files/8240896/artisan-settings-stevek-v4.0b.aset.txt)

MAKOMO, 2022-03-14:
A pity you did not test our 2.4.7 beta to have us iron this out before the release. I found the culprit and a second problem. Both should be solved now. Could you confirm by checking the current Continuous build: https://github.com/artisan-roaster-scope/artisan/releases/tag/continuous

Brewzologist, 2022-03-14:
Yah, sorry.  I wanted too but have been a bit busy lately.  I will try the continuous build in the next day or so and provide feedback.  Glad you were able to quickly find the issue.

Brewzologist, 2022-03-19:
Marko;  I tested my ramp/soak profiles using the continuous build with file dates of 14-Mar-2022 and build number below.  They are working properly again, so your fix appears to be solid.  Well at least on my roaster! :)  Thx!

Artisan Version [e.g. v2.4.4]: v2.6.0
Artisan Build (number in brackets shown in the about box): 0eae72e

MAKOMO, 2022-03-20:
Thanks for testing and this confirmation!

closed 2022-03-20 by MAKOMO