artisan-roaster-scope/artisan #850: 'NoneType' object has no attribute 'poll' error in 2.6.0 connecting to Aillio Bullet R1

stuartmcknight, 2022-03-16:
**Describe the bug**
'NoneType' object has no attribute 'poll' error when hitting ON after relaunching Artisan after initial config.

**To Reproduce**
Steps to reproduce the behavior:
Launch Artisan 2.6.0 - factory reset config. Choose Aillio Bullet R1 IBTS with roaster connected. Hit ON, scope monitor message is replaced with R1 status of OFF. Buttons work to control roaster and data on temperature is displayed.

Make some changes in Artisan to setup (change to C from F etc). Quit Artisan and relaunch.
When now hitting ON button, error message about is displayed and no control or data from the R1 is available.

**Expected behavior**
Normal control of R1.

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Setup (please complete the following information):**
 - Artisan Version [e.g. v2.4.4]: 2.6.0
 - Artisan Build (number in brackets shown in the about box) [e.g. 97e6dec]: 
 - Version [e.g. 22]: 
 - Platform (Mac/Windows/Linux + OS version) [e.g. macOS 10.15.3]: Windows 10
 - Connected devices or roasting machine [e.g. Probatone 5]: Aillio Bullet R1 IBTS

**Additional context**
Add any other context about the problem here.

Please attach your current Artisan settings file (as exported via menu Help >> Save Settings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._
[artisan-settings copy.txt](https://github.com/artisan-roaster-scope/artisan/files/8272238/artisan-settings.copy.txt)

MAKOMO, 2022-03-19:
Stuart, do I understand this correctly, that initially Artisan connected successfully and did show data and allowed control, but after the restart it did not? Is this a sporadic issue or does Artisan now fail constantly  to connect to the machine? Does the error message also contains a line number?

Could you please attach also the artisan.log file (click on the plus icon while holding the OPTION (macOS) or ALT (Windows) key), rename the file to artisan.log.txt and attach it here via drag-and-drop directly to a Github comment field.

stuartmcknight, 2022-03-19:
Hi Marko, yes it seems that if I reset the configuration, then choose Bullet R1 IBTS things connect and both data and control work. On quitting and restarting it seems that error comes back (no line number that I have ever seen) until another reset. This is of course problematic if you have made changes like temperature units etc.

I will get the logs and see if I can expand the repro steps shortly.

stuartmcknight, 2022-03-19:
[artisan.log](https://github.com/artisan-roaster-scope/artisan/files/8309804/artisan.log)

MAKOMO, 2022-03-26:
Stuart, could you please give us a more detailed step-by-step guide which allows to reproduce this issue (whatever you do after reset)? 

Could you please also tell as if Artisan fails constantly to connect to the machine after this reset + those configuration changes? What exactly are the symptoms but for that error message?

As you know there is currently no developer with access to an R1, however, we might have the chance to play with a unit for some hours. It would be good to have precise idea of what is going wrong and how to reproduce in order to allow us to fix this, whatever it is. 

Thanks from the whole Artisan team for your reports!

stuartmcknight, 2022-03-28:
Hi Marko,

Sorry for the delay. I have tried to narrow down the possible causes. I tried another Windows 10 laptop which had never had Artisan or the Aillio drivers before and the same pattern emerged where an initial configuration worked but quitting Artisam and relaunching it gave the error. It seems that if I do a factory reset then select the Bullet in the machine menu, I can retore communication but I will do some more testing. As you know the drivers have to be the old libusb-win to work with the Bullet although Aillio have moved on to the winusb driver. I have been using a itility called Zadig to install and switch between them so I could use either RoasTime or Artisan. This switching worked prior to 2.6.0 - I guess I could go back to 2.4 (?) and verify. I'll also try installing the legacy driver from Sweet Marias which should be the same.

Also trying to get my hands on a MacOS laptop to try; this might help confirm if it is a Windows/Windows Driver related issue.

I don't know if while you have access to the Bullet machine, someone might be able to look into changing the driver over to the most recent winusb (I think there is still an enhancement request on that).

I will of course update should I get more information.

stuartmcknight, 2022-03-28:
Well very odd Marko but I can't reproduce this again. I used Zadig again to install the driver (you may want to update your documentation as it appears Sweet Maria's no long list the legacy driver on their site for download and your link gives a 404 error). I experimented adding one config change at a time and quit/restarted Artisan in between. There were a couple of sessions where the message said "scope monitoring..." without an update but in these cases, the data for the temperatures etc was still valid from the Bullet and the control seemed to work. I even played back a background roast and it worked for the most part (there were a couple of fan change events that it seemed to ignore but I spotted them and fixed it manually).

I'll continue investigation but maybe this was a glitch in the matrix.

MAKOMO, 2022-04-10:
That error indicates that Artisan cannot access the USB port at the time of the error. Initial experiments by Dave indicate that this is caused sometimes by swapping serial drivers with Zadig, but it is still unclear how exactly to reproduce. In any case, this seems to be external to Artisan. Dave is still investigating....

stuartmcknight, 2022-04-11:
I agree it's a puzzler because resetting the configuration shouldn't fix a driver issue. I had updated to the latest Zadig release  recently so not sure what impact that might have had. If I come across any new data, I will of course share. For now I don't want to waste any more of your valuable time. Unless inquiring minds just have to know. I suffer from that too ;-)

roasterdave, 2022-04-16:
Stuart, I was given some time on a Bullet R1 V2 and found several things.  Occasionally the libusb driver is not properly initialized after Zadig switches from WinUSB to libusb-win32.  As a result Artisan encounters a problem talking to the driver and shows that  `'NoneType' object has no attribute 'poll'` error.  Other errors are possible as a result of the driver not functioning.  A Windows restart fixes it 100% of the time.  Other methods of "kicking" the driver might work too.

More interesting, I found that RT talks to the Bullet just fine on libusb.  Use Zadig to switch the driver from WinUSB to libusb-win32.  In my tests on Windows it runs just fine.  Sometimes when RT starts it will put up a driver warning that can be ignored because the Green Light is on in the lower left corner and everything works.  You can't run both RT and Artisan at the same time (and never will) but in my tests there was no need to flip the driver back and forth.  I ran repeated back-to-back roasts alternating between Artisan and RT with no issue. This was with RT v3.4.1 and firmware v602.  Would you confirm this method works for you?

stuartmcknight, 2022-04-18:
Hi Dave

Thanks for the follow up - really RoasTime 3.4.x still reads the old 
drivers ? Must give that a try - didn’t think that worked last time, 
but maybe I got turned around in testing. I do have the 602 beta 
firmware on my roaster.

You’re right, I had forgotten how Windows loves to restart even if you 
just want to delete a printer ;-)

Best

Stuart

roasterdave, 2022-05-02:
Hi Stuart,

Have you tried running RT with libusb?  Does it work properly for you?

stuartmcknight, 2022-05-09:
Hi Dave,

Sorry for the delay getting back to you.

Well what do you know, I thought I was using the RT driver to connect to 
my Bullet but according to Zadig and the Driver Manager, it is 
libusb-win (guess I must have forgotten to switch over or maybe Windows 
just messing with me).

So you are totally spot on - you can switch between RT and Artisan (so 
far) just by quitting and launching the one you want.

Wonder how long before that stops working…

Stuart

roasterdave, 2022-05-12:
Hi Stuart,

Thanks for 'accidentally' confirming!  We'll keep fingers crossed this continues to be allowed to work.

roasterdave, 2022-05-12:
Instructions  are updated on the [Machines](https://artisan-scope.org/machines/aillio/)  page.

closed 2022-05-12 by roasterdave