artisan-roaster-scope/artisan #858: Control Button - Color Change when turned on

chuckcoffee, 2022-03-30:
Hi,

The PID control feature is awesome. I am using that for the BPP. Really helps stabilize my Huky Roaster both for startup and between batches.
Currently when the PID is turned on the "Control" button will change to a slightly lighter blue. The change is pretty small and I often cannot tell and have to open the PID screen to and click on "on" to verify.

Can there be a choice for the user on what colors for both on and off.

MAKOMO, 2022-05-14:
Chuck, which version of Artisan are you using. This has been already addressed in v2.6.0. See below how things look for me on macOS v2.4.6 vs 2.6.0. Already in 2.4.6 I could clearly differentiate those states. To me those looked already like black vs white. Still we decided to render the ON state even lighter. How does this look on your side?

v2.4.6:

[image]

v2.6.0:

[image]

chuckcoffee, 2022-05-14:
MAKOMO
I am on ver 2.6.0. I can see the difference when they are side  by side. In use though I am are looking at only one at a time so have nothing to compare it to. I roast outside so the differences are also a bit more muted so I was looking for something more distinct. Either the option for the user to set the fill/font color or if hard coded I am showing some possible options in the image below. 

Consider if I was writing a work instruction that rather than "when the PID Control is on the fill is a lighter blue " it would say "when the PID Control is on the font will turn red". My 2c

[image]

MAKOMO, 2022-05-17:
I modified the PID button active color to be red in Commit 51fb0ab05e0a6d6666f0f5a239594b4321d106ac although to me this is breaking our design rules. Please test the current continuous build.

chuckcoffee, 2022-05-17:
Ok, I tested the current continuous build. Works great. Thank You!
[image]