artisan-roaster-scope/artisan #859: 2.60 and 2.61 and lag at the DROP moment

procent85, 2022-04-02:
**Describe the bug**
Hello. In version 2.60 and 2.61 I am experiencing a lag at the DROP moment. 

**To Reproduce**
Steps to reproduce the behavior:
The earlier stages of roasting the coffee are fine, but at the end when I open the lid and throw the beans into the cooling tray, the DROP temperature and time are marked with a delay. Fof example: I dropped my coffee at 26% DEV, 12:50 min. and 208.0 C but Artisan marked DROP as 27%, 12:54 and temp. 203 C. I attached a high resolution screenshot. Please zoom in to the end of the graph, you can see the delay with which Artisan "marked" the drop.

**Expected behavior**
Marking DROP without lag as in previous versions.

**Screenshots**
Please see the attachment.

**Setup (please complete the following information):**
 - Artisan Version 2.60 and 2.61
 - Platform Windows-10
 - Connected devices or roasting machine: Coffed SR5 manual

Please attach your current Artisan settings file (as exported via menu Help >> Save Settings as *.aset) file.
[image]
[KOL NR2 4.7KG 26 I 207.4 ST - MAI 3_51 - CH 205 G 0_40 - TP 78 DE 5_37.txt](https://github.com/artisan-roaster-scope/artisan/files/8402252/KOL.NR2.4.7KG.26.I.207.4.ST.-.MAI.3_51.-.CH.205.G.0_40.-.TP.78.DE.5_37.txt)
[artisan-settings - 2.6.1 with lag.txt](https://github.com/artisan-roaster-scope/artisan/files/8402253/artisan-settings.-.2.6.1.with.lag.txt)

Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

procent85, 2022-05-19:
I would like to add that I have recently formatted my SSD drive, installed Windows 10 with the latest updates and did a clean installation of Artisan . Unfortunately, there is still lag on my Coffed SR5 machine at the DROP moment. I am posting a high res. screenshot from today's roast. I finished roast at 207.0 C and 12:27 min. 
[image]

MAKOMO, 2022-07-05:
Could you please test the current development build v2.6.1 if that one performs any better and report here?

https://github.com/artisan-roaster-scope/artisan/releases/tag/continuous

procent85, 2022-07-12:
Hi, after a dozen roasts I can say that this bug has been fixed.

MAKOMO, 2022-07-13:
Thanks for this confirmation

closed 2022-07-13 by MAKOMO