artisan-roaster-scope/artisan #864: Add another format ~dCHARGE in Events Annotations

Nicodmk, 2022-04-04:
"~dCHARGE" function in Annotations only show number of seconds from Charge, Can you make a new different format for "minutes:seconds" from Charge, maybe it will be easier to read without making calculation when roasting or reading artisan profile. 
Thank you.

roasterdave, 2022-04-10:
Good idea.  This will be in the next release.  Until then it is available in the [continuous builds](https://github.com/artisan-roaster-scope/artisan/releases/tag/continuous).  Click the Help button in the events annotations dialog to see the new fields.

closed 2022-04-10 by roasterdave

Nicodmk, 2022-04-21:
I have tested "~dCHARGE_ms" mode in this continuous builds a few days in Linux , so far so good. Thanks for the update.