artisan-roaster-scope/artisan #869: File name becomes empty  when it's in Chinese.

figoluo, 2022-04-20:
**Describe the bug**
When I type the title in Chinese, the auto-save file's name would be empty.
A clear and concise description of what the bug is.

**Expected behavior**
Hope the file's name can be auto saved in Chinese.

**Screenshots**
[image]
[image]

**Setup (please complete the following information):**
 - Artisan Version [v2.6.0]:
 - Platform[macOS 12.0.1]:

MAKOMO, 2022-05-02:
Artisan never supported no-ascii chars in filenames. We plan to add support for general unicode characters to the next release. Thanks for the suggestion.

MAKOMO, 2022-05-13:
Support for filenames with unicode characters has been added in Commit fc50b49c8e815293c6851564cb8f7486b952413c. 

This is available in the beta version 2.6.1 which is available here:
https://github.com/artisan-roaster-scope/artisan/releases/tag/continuous

Could you please test this and give us feedback?

ID-JerryChen, 2022-06-25:
Makomo Thank you, the result of my test here is ok
[image]

ID-JerryChen, 2022-06-25:
There is another error
cause the text to not display properly

[image]

MAKOMO, 2022-07-14:
Thanks for testing/confirming the original issue and reporting also this new issue. The monospace font used here did not feature Chinese characters. Just replaced the font here such that this should display fine now also with Chinese characters.