artisan-roaster-scope/artisan #870: Feature request: add data for Rubasse Roaster

loveggrex, 2022-04-22:
I see that you have support for the Rubasse roaster.

Just wondering that if possible for showing more data, such as air pressure, drum temperature & rotation, airflow temperature...etc

If you have any interest in adding support for that, we're happy to assist any programmer working on it.

Thanks
Rex F.

MAKOMO, 2022-04-23:
Can we get any information on the machines communication protocol to connect it?

MAKOMO, 2022-05-14:
That CSV file does not describe which column holds which data as there are no headers included. Current Artisan is already importing the data from the first 9 columns, you just need to make those corresponding extra device channels in Artisan visible (Config >> Device, 3rd tab).  The remaining columns seem to hold data of a background/template profile. Artisan is not importing data from two profiles from one file.

Further that CSV file seem not to correspond to the data shown in the screenshot. The moisture in the CSV file is always zero and I could not locate the Drum Temperature data. Any help with a proper description of this data set (incl. data ranges) would be helpful.

loveggrex, 2022-05-16:
We really appreciate your help and support.

I'll check with our software engineer and get back to you.

Sincerely,

-- 
Rex Fan
Sales Consultant

Rubasse Co. Ltd.
+886 975631798 <+886975631798> | Whatsapp, Wechat, Line
***@***.***
www.rubasseroasters.com
[image: facebook] <https://www.facebook.com/rubasseroasters/>
[image: instagram] <https://www.instagram.com/rubasseroasters/?hl=zh-tw>

MAKOMO, 2022-06-18:
Any news on this?

MAKOMO, 2022-07-05:
Stall

closed 2022-07-05 by MAKOMO

loveggrex, 2022-07-06:
Sorry for the late reply, the project is still pending.
I will keep you updated once we get progress.

Thank you.

-- 
Rex Fan
Sales Consultant

Rubasse Co. Ltd.
+886 975631798 <+886975631798> | Whatsapp, Wechat, Line
***@***.***
www.rubasseroasters.com
[image: facebook] <https://www.facebook.com/rubasseroasters/>
[image: instagram] <https://www.instagram.com/rubasseroasters/?hl=zh-tw>

MAKOMO, 2022-07-06:
Could you please give a timeline for this. We want to go on. Thanks.