artisan-roaster-scope/artisan #881: Modbus communication error MACBOOK M1

MrBlend, 2022-05-11:
Hello!
I'm having temperature readout issues. 

Computer: MacBook Air M1
System: Monterey 12.3.1
Artisan version: tried 2.6.1. and 2.6.0.
Roaster: Coffed SR5 manual
Using also the newest Modbus driver: CH34xVCPDriver

Roaster is connected via usb cable to port USB --> USB-C

The port is ok, can plug there literally anything and it's working. Roaster itself works great with Win10 - no problems with temperature readouts. On MacBook Air M1 it opens as normal but once I go "start" it shows 0 on BT and any other temperatures. Artisan also shows the message "Modbus communication error". I have chosen right roaster, tried different ports, read everything what I can find on-line and it's still not working.

Have no idea what to do - have you experienced similar issues with new MacBooks Air M1? Is there any solution or new my laptop can't be used for roasting?

Will be REAAAAALLLLYYYY grateful for any tips to try :)

Many thanks!
Paweł

MAKOMO, 2022-05-12:
Sounds like a USB driver issue to me, external to Artisan. The Artisan team does not has access to any M1 machine nor a Coffeed machine.  Anyone else? Maybe Coffed can support you here.

MrBlend, 2022-05-18:
After a week and many driver instalation attempts, new cables etc. the solution was much easier...
Maybe someone will find it useful :)
[image]

I just copy-pasted all port data from Modbus section and it's running (jpg above config--->ports). Many things were different, So easy... Stupid me :)

Cheers!
Pawel

closed 2022-06-19 by MAKOMO