artisan-roaster-scope/artisan #882: Exception with VA18B

bowsi, 2022-05-12:
When using a VA18B multimeter the error "a bytes-like object is required, not 'int' "
occurs in function TEVA18Btemperature in source file comm.py at this line:
                    s201 = binascii.hexlify(r[1])
I have found that if this line (and the following eight or so similar lines) is changed to
                    s201 = binascii.hexlify(r[1].to_bytes(1,'big'))
then the exception goes away and the temperatures are correctly read.
I'm using PyCharm 2021.2 on Windows to debug this but first found the error when I installed Artisan on a linux machine recently.

[image]

MAKOMO, 2022-05-13:
Excellent contribution! I just applied your fix to the current trunk. Could you please test if the current continuous build works for you as expected?

https://github.com/artisan-roaster-scope/artisan/releases/tag/continuous

bowsi, 2022-05-13:
Thanks Marko. I have tested and can confirm that the problem is now fixed.

MAKOMO, 2022-05-13:
Wow! That was fast. Thus I close this issue. Thanks again for your fix.

closed 2022-05-13 by MAKOMO