artisan-roaster-scope/artisan #885: Support Windows Package Manager (winget) as a delivery and update mechanism

josarsepi, 2022-05-16:
**Is your feature request related to a problem? Please describe.**
It is difficult for users to maintain 3rd party software on Windows OS, this includes Artisan.

**Describe the solution you'd like**
Windows Package Manager (winget) is a CLI based tool for managing packages on MS Windows.  I would like to be able to 1) search for and find Artisan, 2) install Artisan, and 3) update Artisan using winget.

**Describe alternatives you've considered**
- Using linux - I do, just not everywhere I use Artisan.  I'm sure I'm not alone here.  I'm a huge fan of linux but sometimes I just want to use Windows.  Also, it's not in the all the distro repositories (or any?) so this process would be the same as the manual installation for Windows OS.
- Using chocolaty - I do, but choco doesn't provide the centralized package repository that is available in winget.
- Going out to the repo and checking for updates and downloading then installing those updates manually - That's what we all have to do today.

**Additional context**
Windows Package Manager (winget) does not support .zip format so the current releases can not be used.  If releases for Windows were made available as .exe, a manifest could be developed that allowed for the use of Windows Package Manager for finding, installing, and updating Artisan software:
- `winget search artisan`
- `winget install artisan`
- `winget upgrade artisan`
- `winget upgrade --all`

roasterdave, 2022-06-11:
It looks possible.  This seems to be a little known feature of Windows.  Will consider this for the next major release.  Thanks for the suggestion.

roasterdave, 2022-09-27:
Unfortunately Artisan is not able to be added to the Winget repository.  Making an entry requires signing the Microsoft Contributor License Agreement (CLA).  This licence is not friendly.  It requires assigning rights and making representations that are neither possible nor acceptable.  Other parties have raised these concerns too.  Microsoft states they are not currently considering any changes to their CLA process. For now we must pass.

closed 2022-09-27 by roasterdave