artisan-roaster-scope/artisan #892: ModbusTCP timeout settable

tofublock, 2022-05-31:
I'm implementing a client device for Artisan as a ModbusTCP slave and sometimes my device doesn't respond quite fast enough for Artisan, as its read timeout is fixed to be min(0.2s, samplingtime/2). Setting the timeout to something like 0.5s fixes this problem for me and communication works well with fewer interruptions.

Allowing the timeout for ModbusTCP to be set from the "Config>Port" dialog would fully solve my problem. The timeout field that is there is currently only used for SerialModbus connections, but could be used for the same purpose without the need to change anything else.

tofublock, 2022-05-31:
https://github.com/artisan-roaster-scope/artisan/blob/075b54cd6ac1ee42730764f0c48eae134d1b428f/src/artisanlib/modbusport.py#L269
and
https://github.com/artisan-roaster-scope/artisan/blob/075b54cd6ac1ee42730764f0c48eae134d1b428f/src/artisanlib/modbusport.py#L287
both just need to be changed to `timeout=min((self.aw.qmc.delay/2000), self.timeout) # the timeout should not be larger than half of the sampling interval`

MAKOMO, 2022-06-21:
Added an explicit MODBUS IP (TCP/UDP) timeout (and retry) parameter to the MODBUS tab for a cleaner solution to reusing the MODBUS serial timeout. Will close your PR as well. Does this work for you?

tofublock, 2022-06-22:
That's an even better solution, of course. Thank you. I can't test it at the moment but will get back to you once I can.