artisan-roaster-scope/artisan #898: pdf filename been cutoff while saving after roast

cavixyeh, 2022-06-16:
**Describe the bug**

whenever I change or update the details in roast data and try to save the file, the save function for also-save in other filetype using PDF  always keeps the name only the first word before "." , eg: "Col.pdf" not "Col.XXXestate.W.500g_22-06-16_1400" originally while roasting and click OFF for saving file. Even the existing pdf file had been deleted by myself earlier.

 
**To Reproduce**
Steps to reproduce the behavior:
1. Open alog file and do anything for update infos
2. Click  "save" or "save as" or "keep a copy as"
3. See error

**Expected behavior**
the file name won't be cut after "." like normal saving when finish the roasting. 

**Screenshots**
[image]
[image]

**Setup:**
 - Artisan Version [e.g. v2.4.4]: v2.4.2
 - Artisan Build (number in brackets shown in the about box) [e.g. 97e6dec]:   build 3 (bcaf44d) 
 - Version [e.g. 22]: where to find it??
 - Platform (Mac/Windows/Linux + OS version) [e.g. macOS 10.15.3]:  Mac OS High Sierra 10.13.6
 - Connected devices or roasting machine [e.g. Probatone 5]: Kapok 1.0

**Additional context**
[cavix-2022066.aset.txt](https://github.com/artisan-roaster-scope/artisan/files/8916905/cavix-2022066.aset.txt)

Add any other context about the problem here.

Please attach any relevant Artisan *.alog profiles.
[Hon.PedroOrchid.N.500g_22-06-08_1520.alog.txt](https://github.com/artisan-roaster-scope/artisan/files/8916901/Hon.PedroOrchid.N.500g_22-06-08_1520.alog.txt)

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

roasterdave, 2022-06-16:
Thanks for reporting!  It is confirmed in v2.6 and will be fixed in the next release.  

Looks like you are using a legacy version of macOS.  Artisan v2.6 includes a release build for legacy macOS, however the continuous build platform no longer supports them.  Support of legacy platforms is uncertain for future releases.  

If you must stay on the legacy OS then to avoid this issue you might consider using a character other than '.' in your file names.

closed 2022-06-16 by roasterdave

cavixyeh, 2022-06-16:
> Thanks for reporting! It is confirmed in v2.6 and will be fixed in the next release.
>
> Looks like you are using a legacy version of macOS. Artisan v2.6 includes a release build for legacy macOS, [...]

thank you!!