artisan-roaster-scope/artisan #907: BT curve and ROR lag

bolza4, 2022-06-27:
Hello everyone, 

I'm having a hard time tracking the BT curve and the ROR curve on the background. 

It seems like I can only align with the BT curve or the ROR curve, in other words I have to choose if I wanna track one curve or the other.
If I track the BT curve, I have a delay on the ROR curve, or the other way around.

How can I fix it?
thank you so much :)

[image]

[image]

matteo

MAKOMO, 2022-07-03:
Please attach your smoothing settings here. Thanks!

bolza4, 2022-07-03:
> 

there you go! :) thanks.

[image]

bolza4, 2022-07-05:
would you say that this smoothing settings are causing time shifts? thanks a lot :)

bolza4, 2022-07-05:
> Please attach your smoothing settings here. Thanks!

would you say that this smoothing settings I posted above are causing time shifts? thanks a lot :)

MAKOMO, 2022-08-15:
Yes. This is caused mostly by the strong curve smoothing here which is not applied to the foreground profile during recording. Turn that to 0 or 1 and the shift is gone. Better use directly a higher delta span for RoR smoothing.

We will limit the amount of curve smoothing in the next version and might remove it completely in a later version also due to those complications.

closed 2022-08-15 by MAKOMO