artisan-roaster-scope/artisan #910: Artisan 2.6.0 PID control messages for TC4 hardware PID

hiroyuki2080, 2022-06-27:
**Describe the bug**
A clear and concise description of what the bug is.

There are 2 issues - the first one is not a realy an issue on TC4 but the second one affects to the ramp stage.
First issue
   At the first batch and PID start, PID control sends out PID parameters and SV in one block(see serial monitor below).
  This happens only at the first batch of Artisan starts up.
Second issue
   PID control sends out PID parameters and SV is coming up in the following second but SV value starts from very low temperature(see serial monitor below)
############
##First batch##
############
# Active channels set to 1200
UNITS;C
FILT;70,70,70,70
READ
27,213,210,0,0,0.01
READ
27,213,210,0,0,0.01
ot2 50
ot1 50
READ
27,213,210,0,0,0.01
READ
27,213,211,0,0,0.01
PID;T;0.1;0.5;0.1
PID;CHAN;1
PID;CT;1000
PID;LIMIT;0;100
PID;ON
PID;SV;159.0                < SV in the same block with PID parameters
READ
27,213,211,0,0,0.01
PID;SV;4.72                  <SV starts from very low temperature
READ
27,214,211,0,0,0.01
PID;SV;6.5
READ
27,214,211,0,0,0.01
PID;SV;8.27
PID;OFF
CHAN;1200
# Active channels set to 1200
UNITS;C
FILT;70,70,70,70
READ
27,214,211,0,0,0.01
READ
27,214,211,0,0,0.01
ot2 50
ot1 50
READ
27,214,211,0,0,0.01
READ
27,214,211,0,0,0.01
READ
27,214,211,0,0,0.01
PID;T;0.1;0.5;0.1
PID;CHAN;1
PID;CT;1000
PID;LIMIT;0;100
PID;ON
READ
27,214,211,0,0,0.01
PID;SV;6.81             <SV starts from very low temperature
READ
27,214,212,0,0,0.01
PID;SV;8.58
READ
27,214,212,0,0,0.01
PID;SV;10.36
PID;OFF

Best regards
Hiroyuki Yamamoto

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Setup (please complete the following information):**
 - Artisan Version [e.g. v2.4.4]:  v2.6.0
 - Artisan Build (number in brackets shown in the about box) [e.g. 97e6dec]: 7088095
 - Version [e.g. 22]:
 - Platform (Mac/Windows/Linux + OS version) [e.g. macOS 10.15.3]: Windows 10
 - Connected devices or roasting machine [e.g. Probatone 5]:  TC4

**Additional context**
Add any other context about the problem here.

Please attach your current Artisan settings file (as exported via menu Help >> Save Settings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._
[image]
[image]
[artisan-settings.aset.txt](https://github.com/artisan-roaster-scope/artisan/files/8994330/artisan-settings.aset.txt)

MAKOMO, 2022-07-03:
I don't understand the first part of the issue. TC4 commands are not send in blocks, but as individual commands. The initial SV after initializing the TC4 PID is just part of that initialization.

The low SV might be due to a bug in the Ramp/Soak processing within Artisan. I have to check.

hiroyuki2080, 2022-07-04:
Thank you for looking into the issue. The first one is not really a matter but the second issue makes higher ramp speed than the one starting from a normal current temperature after a warming period.

MAKOMO, 2022-07-05:
I made a modification to the code such that the SV is not initialized on PID initialization if in Ramp/Soak mode.

Regarding the second issue: the SV Ramp/Soak pattern starts always with SV 0. Thus it is expected that after turning the PID on the RS SV starts from very low target temperatures.  However, it will reach the target temperature within the 2min as defined by your first RS item. See attached image.

To start from a different SV you would introduce a first RS item with just a Soak part at minimal duration (00:01) to set your starting SV.

Does this help?

[image]

[image]

closed 2022-07-07 by hiroyuki2080