artisan-roaster-scope/artisan #914: Arduino UNO R3 with 3 AdafruitMAX31856 Shields - How to make it work

thedimou, 2022-07-07:
Hi there,
First of all thank you for contributing in our coffee society.

Now to the point I have programed an arduino uno board with 3 Max31856 shields and getting temps both in Arduino IDE and PlatformIO (VS Code).

How could I make it communicate with artisan? 

Here is my code for the arduino uno board.
`
#include <Adafruit_MAX31856.h>
#include <Arduino.h>

Adafruit_MAX31856 maxthermoET = Adafruit_MAX31856(4, 8, 9, 3);
Adafruit_MAX31856 maxthermoBT = Adafruit_MAX31856(5, 8, 9, 3);
Adafruit_MAX31856 maxthermoAT = Adafruit_MAX31856(6, 8, 9, 3);

void setup() {
  Serial.begin(115200);
  while (!Serial) delay(10);
  Serial.println("MAX31856 thermocouple test");

  if (!maxthermoET.begin()) {
    Serial.println("Could not initialize thermocouple.");
    while (1) delay(10);
  }

  if (!maxthermoBT.begin()) {
    Serial.println("Could not initialize thermocouple.");
    while (1) delay(10);
  }
  
  if (!maxthermoAT.begin()) {
    Serial.println("Could not initialize thermocouple.");
    while (1) delay(10);
  }

  maxthermoET.setThermocoupleType(MAX31856_TCTYPE_K);
  maxthermoBT.setThermocoupleType(MAX31856_TCTYPE_K);
  maxthermoAT.setThermocoupleType(MAX31856_TCTYPE_K);

/*  Serial.print("Thermocouple type: ");
  switch (maxthermoET.getThermocoupleType() ) {
    case MAX31856_TCTYPE_B: Serial.println("B Type"); break;
    case MAX31856_TCTYPE_E: Serial.println("E Type"); break;
    case MAX31856_TCTYPE_J: Serial.println("J Type"); break;
    case MAX31856_TCTYPE_K: Serial.println("K Type"); break;
    case MAX31856_TCTYPE_N: Serial.println("N Type"); break;
    case MAX31856_TCTYPE_R: Serial.println("R Type"); break;
    case MAX31856_TCTYPE_S: Serial.println("S Type"); break;
    case MAX31856_TCTYPE_T: Serial.println("T Type"); break;
    case MAX31856_VMODE_G8: Serial.println("Voltage x8 Gain mode"); break;
    case MAX31856_VMODE_G32: Serial.println("Voltage x8 Gain mode"); break;
    default: Serial.println("Unknown"); break;
  }*/

  maxthermoET.setConversionMode(MAX31856_ONESHOT_NOWAIT);
  maxthermoBT.setConversionMode(MAX31856_ONESHOT_NOWAIT);
  maxthermoAT.setConversionMode(MAX31856_ONESHOT_NOWAIT);

  delay(1000);

}

void loop() {
  
  maxthermoET.triggerOneShot();
  maxthermoBT.triggerOneShot();
  maxthermoAT.triggerOneShot();
  
  delay(1000);
  
  if (maxthermoET.conversionComplete()) {
    Serial.print(maxthermoET.readThermocoupleTemperature());
    Serial.print(",");
  } else {
    Serial.println("Conversion not complete!");
  }

  if (maxthermoBT.conversionComplete()) {
    Serial.print(maxthermoBT.readThermocoupleTemperature());
    Serial.print(",");
  } else {
    Serial.println("Conversion not complete!");
  }

  if (maxthermoAT.conversionComplete()) {
    Serial.print(maxthermoAT.readThermocoupleTemperature());
    Serial.print(",");
  } else {
    Serial.println("Conversion not complete!");
  }
}
`

Thank you in advance.

mrpenner, 2022-07-09:
This is the way I did it in my [program](https://gitlab.com/mrpenner/roaster-display/-/blob/master/roaster-display.ino):
```
    if (Serial.available() > 0 && Serial.readStringUntil('\n') == "READ") {
        Serial.print(AT, 2);
        Serial.print(',');
        Serial.print(ET, 2);
        Serial.print(',');
        Serial.print(BT, 2);
        Serial.print(',');
    }
```
That will send the temperatures when Artisan requests them. [Here](https://artisan-scope.org/devices/arduino/) is some info about using an Arduino with Artisan. If you want to emulate more of the TC-4 functionality, take a look at the [commands](https://github.com/greencardigan/TC4-shield/blob/master/applications/Artisan/aArtisan/trunk/src/aArtisan/commands.txt) it recognizes.

Then in Artisan go to Config>Device... and select TC4, and go to Config>Port... where you will need to change the Comm Port to the port you have your Arduino connected to, and the rest I think can be the same as my Port settings:
[image]

thedimou, 2022-07-11:
Thank you so much, I figured a way out, which I'm going to upload in the next couple of days, but there is always for improvement. The commands are going to be super useful by the way. Appreciate.

closed 2022-07-14 by MAKOMO