artisan-roaster-scope/artisan #920: Asian plus document upload issue

elguapo1911, 2022-07-21:
I was in artisan plus adding recently purchased coffeebeans. I normally upload the PDF with the organic certification but it won't let me upload a PDF format? I swear I've done this before. Any suggestions?

MAKOMO, 2022-07-21:
While not in edit mode of the new bean entry, just click on the "Upload documents" link to open the Documents Uploader widget which allows you to select a set of documents to be attached via the "+".

[image]

Hope this works for you!

elguapo1911, 2022-07-21:
I'll give it a try out of edit mode. Thanks!

MAKOMO, 2022-07-21:
Oh, no. You should be "NOT" in edit mode.

elguapo1911, 2022-07-21:
Lol ok, my bad

closed 2022-08-24 by MAKOMO