artisan-roaster-scope/artisan #947: Any possibility to work with Tasi TA612C four channel thermometer with USB

Mikekkc, 2022-08-30:
## Expected Behavior
Collect data from Tasi TA612C ( http://www.china-tasi.com/plus/view.php?aid=97 )

## Actual Behavior
Construct the roasting curve with channels on BT, ET and third reading like ambient temperature

## Steps to Reproduce the Problem

  1. Windows connected
  2. Data not read by Artisan
  3. No curve generated

## Specifications

  - Artisan Version: 2.6
  - Artisan Build (number in brackets shown in the about box):
  - Platform (Mac/Windows/Linux + OS version): Windows
  - Connected devices or roasting machine: TA612C and SR800

Please attach your current Artisan settings file (as exported via menu Help >> Save Settings as *.aset) file. currently using Victor86B single channel.
Please attach any relevant Artisan *.alog profiles. not available

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2022-08-30:
This device is not supported by Artisan and the team has no resources to add support for it. Please implement the corresponding driver yourself and submit the code as Pull Request for integration.

YuenJay, 2024-01-08:
[TA系列通讯协议.docx](https://github.com/artisan-roaster-scope/artisan/files/13865627/TA.docx)

Here is the communication protocol for the above device. Only Chinese language version.

chihpingkuo, 2024-02-05:
Here is what I've implemented in Rust for reading temperature data from TA612C, hope that helps
https://github.com/chihpingkuo/roastcraft/blob/main/src-tauri/src/devices/ta612c.rs
```
let request: [u8; 5] = [0xAA, 0x55, 0x01, 0x03, 0x03];
self.stream.write(&request);

let mut response: [u8; 13] = [0; 13];
match self.stream.read(response.as_mut_slice()) {
    Ok(_) => {
        let T1 = u16::from_ne_bytes(response[4..6].try_into().unwrap()) as f32 / 10.0;
        let T2 = u16::from_ne_bytes(response[6..8].try_into().unwrap()) as f32 / 10.0;
        let T3 = u16::from_ne_bytes(response[8..10].try_into().unwrap()) as f32 / 10.0;
        let T4 = u16::from_ne_bytes(response[10..12].try_into().unwrap()) as f32 / 10.0;
```

yaserabuzarifa, 2024-11-12:
404 - page not found
Could you please update it?

Thanks

> Here is what I've implemented in Rust for reading temperature data from TA612C, hope that helps https://github.com/chihpingkuo/roastcraft/blob/main/src-tauri/src/devices/ta612c.rs
>
> ```
> let request: [...]

roasterdave, 2024-11-12:
>404 - page not found
Could you please update it?

No one on the Artisan team can help with this request.  Perhaps create an issue at the https://github.com/chihpingkuo/roastcraft repository where the page is missing.

MAKOMO, 2024-11-13:
The file was moved: https://github.com/chihpingkuo/roastcraft-tools/blob/main/rust/src/bin/ta612c_client.rs

closed 2025-12-07 by MAKOMO