artisan-roaster-scope/artisan #950: Modbus Issue Coffee Tech Silon Zr7

MintRoastery, 2022-09-05:
Hello, 

I am trying to figure out a problem I am having with my brand new Silon Roaster, basically I am getting very spotty/laggy temperature readings resulting in my ror spiking all over the place. Its been happing since I got the machine and I am pulling my hair out trying to figure this out. I am getting modbus errors constantly and the problem seems to get worst as the temperature increases, I have attached a profile for example. really at a loss what to do next as I am computer literate but not with modbus unfortunately. Any help would be appreciated .

I initially had Artisan 2.4.4 but updated to 2.6.0 and then 2.6.1 to see if that helped but no change with all versions, also the manufacturer has updated their software on the allen bradley controller.

I am running this on a PC with the latest windows version and connecting directly through cat 6 sheilded ethernet 

[artisan-settings.txt](https://github.com/artisan-roaster-scope/artisan/files/9486770/artisan-settings.txt)

[image]

MAKOMO, 2022-09-05:
To understand what is happening here we need your Artisan log file as we do not have access to that machine. Please do the following:

1) Install Artisan v2.6.0 or even better the current beta version 2.6.1.
2) Hold the ALT+CTRL keys and click on the plus icon in the upper left corner to toggle to the debug logging mode. You should see the message "debug logging ON" in the message line.
3) Record some minutes using Artisan as you would for roasting
4) Hold the ALT key and click again on the plus icon. An email should open with the Artisan log files attached. Please send this message to info@artisan.plus and will take a look.

MintRoastery, 2022-09-05:
Thank you Marko!

I will send that through today

MAKOMO, 2022-09-25:
Never received anything from you. Is this case closed?

MintRoastery, 2022-09-25:
Hi Marko,

No I did, you identified some issues with the MODBUS and instructed me to
change some settings.

Unfortunately it didn't help but I appreciate you looking into for me.
Coffee Tech are sending out some parts that will hopefully address
the issue.

Thanks again

Dion

MAKOMO, 2022-09-25:
Ah right. I lost track... So I will close this here.

closed 2022-09-25 by MAKOMO

s4reea, 2022-10-16:
Hi Dion
if you don't mind to share how was it fixed, what parts were replaced ?
thank you

MintRoastery, 2022-10-20:
Hi Marko,

All I ended up being sent was a wifi modem to connect the roaster to the
computer... However this did not solve the problem. My only solution has
been to bypass their system entirely using my own probes and have connected
to a phidget 1048. I use the control systems on the roaster but ignore all
temperature data coming from the machine as it's just not accurate.

Cheers

Dion Milanesi

s4reea, 2022-10-24:
Thanks for sharing
How can I contact you ?

MintRoastery, 2022-10-26:
Oh is this not Marko?

I'm based in New Zealand... Do you also have a Silon Zr7?

s4reea, 2022-10-31:
> Oh is this not Marko? I'm based in New Zealand... Do you also have a Silon Zr7?
> […](#)

I roast on Ghibli R15 :)

MintRoastery, 2022-10-31:
Are you also having trouble with artisan integration? I find that my
temperature readings from the machines probes are not giving me accurate
readings

s4reea, 2022-11-01:
Not trouble
I figured that you need to use slow sampling rate, 1s is too fast. 

Lets talk on Telegram @s4reea