artisan-roaster-scope/artisan #969: Typhoon 5kg roaster on M1 Apple Macbook - modbus communication Error

deviantworld, 2022-09-26:
**Describe the bug**
trying with different ip addresses as described with the typhoon documentation. But it fails to connect. Error "modbus communication error" 

 - Artisan Version v2.4.6 and 2.6.1
 - Platform Mac OS 12.6. on a M1 Max MBP
 - Connected devices or roasting machine Typhoon 5Kg electric

I tried different ip adresses: 

192.168.0.210/192.168.0.211
no firewall or limit ip address tracking is activated.
Rosetta is installed.

What else can I try?

deviantworld, 2022-09-26:
I think yes. I even tried several Ip addresses the typhoon technician gave me.

MAKOMO, 2022-09-26:
Your mac needs to be on the same IP network as the roaster (with the correct network mask), but using an IP address different to your roaster. In Artisan you need to indicate the IP addres of the roaster.

deviantworld, 2022-09-27:
So I setup the manual ipv4 config in my network settings for the LAN i have plugged in the roaster. I typed in the subnet of 255.255.255.0 and the ip of the typhoon:  198.168.0.211 
also port config at artisan is at the same ip address. 
I dont know what else to do? 
I am using now the 2.6.1 of artisan and it runs on rosetta on the mac M1.

MAKOMO, 2022-09-27:
```
So I setup the manual ipv4 config in my network settings for the LAN i have plugged in the roaster. I typed in the subnet of 255.255.255.0 and the ip of the typhoon: 198.168.0.211
```
This is not a good idea. On an IP network each device on one network has to listen to a different IP address. Thus your computer should be assigned an IP address not used by any other device on that network. If your roasters IP is 198.168.0.211 just use 198.168.0.201 for your computer.
```
also port config at artisan is at the same ip address.
```
This is correct. You need to put the IP address of your roasting machine into the Artisan configuration such that Artisan knows how to address it. Note that Artisan is still running on your computer which needs an IP address different to this (see above).
```
I dont know what else to do?
I am using now the 2.6.1 of artisan and it runs on rosetta on the mac M1.
```
Artisan runs fine on Rosetta and the M1 mac.

deviantworld, 2022-09-27:
Thanks MAKOMO for your advises. so I can use in my network settings just DHCP? Or still manually just with a different ip address as the roaster? 

I tried like you see on the screenshot, but it still says modbus communication error. 

[image]

deviantworld, 2022-09-28:
I tried to find the ip of the roaster, and run several commands in the terminal. But I still dont know which IP the roaster really has: 

jn@Calathea-MBP-M1-Max ~ % ifconfig | grep broadcast
	inet 192.168.0.213 netmask 0xffffff00 broadcast 192.168.0.255
	inet 198.168.0.201 netmask 0xffffff00 broadcast 198.168.0.255
jn@Calathea-MBP-M1-Max ~ % arp -a
? (192.168.0.1) at 38:43:7d:c1:39:f9 on en0 ifscope [ethernet]
? (192.168.0.17) at 12:9d:91:d2:28:e on en0 ifscope [ethernet]
? (192.168.0.210) at (incomplete) on en0 ifscope [ethernet]
? (192.168.0.211) at (incomplete) on en0 ifscope [ethernet]
? (192.168.0.255) at ff:ff:ff:ff:ff:ff on en0 ifscope [ethernet]
? (198.168.0.201) at 0:e0:4c:36:d:c8 on en8 ifscope permanent [ethernet]
? (198.168.0.255) at ff:ff:ff:ff:ff:ff on en8 ifscope [ethernet]
? (224.0.0.251) at 1:0:5e:0:0:fb on en0 ifscope permanent [ethernet]
broadcasthost (255.255.255.255) at ff:ff:ff:ff:ff:ff on en0 ifscope [ethernet]

Anyone any ideas?

MAKOMO, 2022-09-28:
Seems you have your computer on the lan where it can get an IP assigned via DHCP by the router on your network.

I assume you wired the roasting machine to the same network. Maybe that roasting machine, as many others, do not run DHCP and have a fixed IP. In this case you need to connect your PC via ethernet directly to the machine set your IP manually (DHCP deactivated) to one different than the roaster. Thus you build a separate network with just your roaster and your PC on it without any router. That is the standard setup in this coffee roasting world. Only a few machines connect via DHCP.

deviantworld, 2022-09-28:
Alright, so no DHCP. Well as you see in the screenshot I tried most of the time with manual IP and a different IP as the roaster. But still, always the same connection Error. 
How can I ensure what IP address the roaster has? Typhoon says it is 198.168.0.211 but if you do the machine setup within artisan (and use the typhoon preset) it suggests 198.168.0.210

MAKOMO, 2022-09-28:
I would trust the producer of the machine. The suggestion Artisan is displaying is just that, a suggestion which has to be adjusted to fit the concrete machine.

deviantworld, 2022-09-28:
So I tried to ping the ip address of the roaster: 

jn@Calathea-MBP-M1-Max ~ % ping 192.168.0.210
PING 192.168.0.210 (192.168.0.210): 56 data bytes
Request timeout for icmp_seq 0
Request timeout for icmp_seq 1
Request timeout for icmp_seq 2
Request timeout for icmp_seq 3
ping: sendto: No route to host
Request timeout for icmp_seq 4
ping: sendto: Host is down
Request timeout for icmp_seq 5
ping: sendto: Host is down

Also tried with the .211 ip address: 

jn@Calathea-MBP-M1-Max ~ % ping 192.168.0.211
PING 192.168.0.211 (192.168.0.211): 56 data bytes
Request timeout for icmp_seq 0
Request timeout for icmp_seq 1
Request timeout for icmp_seq 2
Request timeout for icmp_seq 3
ping: sendto: No route to host
Request timeout for icmp_seq 4
ping: sendto: Host is down
Request timeout for icmp_seq 5

My ip address in the network is manually set to 
[image]

MAKOMO, 2022-09-28:
What about asking Typhoon for the correct setup?
If your ping does not success, it is unlikely that Artisan can connect.

Anyhow, this is not an Artisan issue thus I closed it here.

closed 2022-09-28 by MAKOMO