artisan-roaster-scope/artisan #982: Issue: Phidget barometer raw data

s4reea, 2022-10-11:
Artisan in ambient readings reads the raw hectoPascals of the phidget for exmple 1010.5hPa
while in phidget control panel I see the actual barometric pressure 90.1k Pa

is this a bug ?

MAKOMO, 2022-10-11:
Did you specify your elevation in the Ambient tab?

s4reea, 2022-10-12:
Yeah
 997 meters

> Did you specify your elevation in the Ambient tab?
Yes
997m

MAKOMO, 2022-10-13:
It is the other way around. The Phidget Control Panel outputs the raw atmospheric pressure which is used by Artisan together with the ambient temperature (defaults to 23C) and the given MASL/hasl to compute the atmospheric pressure using the following function.

```Python
    # computes the barometric pressure from
    #   aap:  atmospheric pressure in hPa
    #   atc:  temperature in Celsius
    #   hasl: height above sea level in m
    # see https://www.wmo.int/pages/prog/www/IMOP/meetings/SI/ET-Stand-1/Doc-10_Pressure-red.pdf
    @staticmethod
    def barometricPressure(aap, atc, hasl):
        return aap * pow((1 - ((0.0065*hasl) / (atc + (0.0065*hasl) + 273.15))),-5.257)
```

Cannot see anything wrong here and the readings I get look fine.

closed 2022-10-13 by MAKOMO

s4reea, 2022-10-13:
[image]

[image]

they are both from same sensor

and when I put at devices amsl 0 height instead of 997
[image]

The right one, So I think we should choose only one settings... phidget or manual height input ?

MAKOMO, 2022-10-13:
I don't understand your question. Sorry.

s4reea, 2022-10-13:
there is difference between actual measured pressure of barometer phidget and the shown value in artisan Roast properties. 

I found the cause of this bug.

If I go to devices -> ambient -> and type the masl there, (997) it shows different pressure (wrong pressure), if I leave it 0, the pressure shown in artisan is right

MAKOMO, 2022-10-13:
This is strange. The pressure Artisan displays needs the correct MASL to be correct as it computes it from the raw atmospheric pressure the sensor is delivering. The Phidget module itself is not able to store a MASL value. So maybe your sensor is measuring wrong and is defect or the value you compare the result with is wrong.

s4reea, 2022-10-13:
I understand what you say. 
I think this thing worths a check
As I have other pressure sensors from Aqara and they are all around 90.5-91kpa